cjcliffe/CubicSDR #190: all the Soapy stuff killed it !

k9wkj, 2015-11-17:
just built off master
and it yelled about not haveing SoapySDR
so i built the soap
then Cubic built (lots of wx warnings)
the device selection dialog comes up  but wont let me use any of the listed devices
could i have done a exclude of the soap somehow?

debian stretch x64

cjcliffe, 2015-11-18:
@k9wkj you'll need to build and install a soapy module as well -- such as SoapyRTLSDR or SoapySDRPlay or SoapyHackRF, etc :)  Binaries with modules included will eventually be available for all platforms once it's out of alpha.

The instructions for RTL-SDR and SDRPlay at least (the rest are similar) can now be found in the wiki:

https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux
https://github.com/cjcliffe/CubicSDR/wiki/Build-OSX
https://github.com/cjcliffe/CubicSDR/wiki/Build-Windows

Cheers!

tomswartz07, 2015-11-18:
Thanks for the info @cjcliffe.

I was unaware of the SoapyRTLSDR build deps as well.

I'll see if I could add in a PKGBUILD to the [Arch AUR repo](https://github.com/tomswartz07/CubicSDR-PKGBUILDS) for these packages.

Is there any thought to how we could streamline the dependency-ness of this project?
Perhaps working with Pothos to see if we could pull their repos as a `git submodule` and integrate them into the CubicSDR build file to compile the other projects automatically?

Dantali0n, 2015-11-18:
@tomswartz07 adding dependencies as sub-modules would make setting things up easier and since everyone working with the project uses the same version of dependencies then, it will make bug tracking easier. Since someone with different versions of the dependencies can get different bugs and errors.

tomswartz07, 2015-11-18:
I agree- that's what I was thinking in suggesting it. :smile: 

Further, we could be sure that end users have all of the tools necessary during build time, rather than pulling from several different locations and/or possibly running things out of the proper order (e.g. building soapyrtlsdr before soapysdr)

guruofquality, 2015-11-18:
I agree, there are a lot of support modules. There used to be less :-) There was a time when I hoped that the support modules themselves would be versioned with the associated hardware's driver package. Anyway, that might mean less soapy modules, but it wouldn't really solve the more general issue of building so many dependencies.

I had some related comments in this issue: https://github.com/cjcliffe/CubicSDR/issues/185 Eventually, we should have more formal linux packaging for many of these dependencies. Covering every distro is a challenge. So, for one reason or another, people are going to be stuck with source installs, but as consumers and not developers of the software -- we should make that easier.

Using cmake with build external project
=====

Rather than submodules, which might get tedious for @cjcliffe to try to keep up to date and version within the repository -- we can make a build system based around cmake that downloads dependencies from git (possibly other places) and builds them with their order of dependencies in mind.

Take a look at PothosSDR, its cmake build amalgamation that uses external project to build my windows installers, its got a lot of stuff in it and stands as a good example. We could easily do something like this for CubicSDR:  https://github.com/pothosware/PothosSDR

* Read more about ExternalProject_Add(): http://www.cmake.org/cmake/help/v3.0/module/ExternalProject.html

Source package based build options
=====

These are systems that take package build descriptions, and download from various sources and build things. I actually have some open issues for some of these systems as well:

* I have an existing homebrew tap, and presumably, it could be used under linux with linuxbrew: https://github.com/pothosware/homebrew-pothos/wiki
* An open issue for arch linux packages, any volunteers @tomswartz07 ? https://github.com/pothosware/SoapySDR/issues/54
* Pybombs is a bit like the others, but one cool thing is that it supports using pre-built packages from apt: https://github.com/pothosware/SoapySDR/issues/51 

Some of these build systems are distributed. So I can host a few repos up on the pothosware github just for the soapy and pothos related projects -- which can be pulled in for cubic as part of the build instructions. But I don't mind expanding said repos to include way more project builds.

However I do want to be able to personally manage the soapy build packages at some level so I can get the packages very complete, dependencies right, and update the git tags and SHA sums when I do releases. Hopefully to make upstreaming the build packages easier and to give users less build problems.

tomswartz07, 2015-11-18:
@guruofquality Great points.

Without further derailing from @k9wkj's original bug, I suggest we further discuss the packaging options in the related issue: #185 

@k9wkj: have you tried installing the SoapyRTLSDR module in addition to the SoapySDR module?

vk2byf, 2015-11-29:
Hi cjcliffe
First of all thank you for sharing your knowledge with us.
My name is Bob VK2BYF and I am a beginner with Linux and this site.
I hope I'm not posting in the wrong place, sorry.
I have followed your instructions for Building CubicSDR for Linux.
I'm using Linux Mint 17.2 Mate.
Everything went well until I got to the part in Build CubicSDR where I run "make"
The error I get is No target specified and no makefile found  STOP it refers to a CMakeOutput.log

It's fairly large file to post here but it basically says sourcefile CMakeCCompileId.c succee$
The output was 0

I carried on to the part called Support Modules SoapyRTLSDR
SoapyRTLSDR seems to already exist and  when run produces Hardware=null and driver=null.

SoapySDRPlay.
fails when I get to cmake .. --DCMAKE_BUILD_TYPE=Release. Cmake error SDRPlay development files not found.

I have tried to faultfind this but no success due to my very limited Linux knowledge. I suspect I may have put something in the wrong directory but I did follow instructions carefully by cutting and pasting commands into the Linux terminal window.

Thanks for taking the time to read this and possibly point me in the right direction.

Regards Bob

cjcliffe, 2015-11-29:
@vk2byf Hi Bob, it sounds like you just need a few more dependencies and are possibly missing a CMake step.

Can you attempt to build CubicSDR again and post here a list of your commands and the output from each and I should be able to spot the issue easily.

Cheers!

vk2byf, 2015-11-30:
Hi Charles, looks like missing dependencies as you said.

Below is what I copied from my terminal as I entered the commands.

starting from my home directory,

*bob@bob-Latitude-E6400 ~ $* git clone
https://github.com/cjcliffe/CubicSDR.git

Cloning into 'CubicSDR'...

remote: Counting objects: 6433, done.

remote: Compressing objects: 100% (119/119), done.

remote: Total 6433 (delta 55), reused 0 (delta 0), pack-reused 6311

Receiving objects: 100% (6433/6433), 52.34 MiB | 465.00 KiB/s, done.

Resolving deltas: 100% (4844/4844), done.

Checking connectivity... done.

*bob@bob-Latitude-E6400 ~ $* cd CubicSDR

*bob@bob-Latitude-E6400 ~/CubicSDR $*

*bob@bob-Latitude-E6400 ~/CubicSDR $ mkdir build*

*bob@bob-Latitude-E6400 ~/CubicSDR $ cd build*

*bob@bob-Latitude-E6400 ~/CubicSDR/build $ *cmake ../

-- The C compiler identification is GNU 4.8.4

-- The CXX compiler identification is GNU 4.8.4

-- Check for working C compiler: /usr/bin/cc

-- Check for working C compiler: /usr/bin/cc -- works

-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - done

-- Check for working CXX compiler: /usr/bin/c++

-- Check for working CXX compiler: /usr/bin/c++ -- works

-- Detecting CXX compiler ABI info

-- Detecting CXX compiler ABI info - done

32 bit compiler detected

-- Looking for XOpenDisplay in
/usr/lib/i386-linux-gnu/libX11.so;/usr/lib/i386-linux-gnu/libXext.so

-- Looking for XOpenDisplay in
/usr/lib/i386-linux-gnu/libX11.so;/usr/lib/i386-linux-gnu/libXext.so - found

-- Looking for gethostbyname

-- Looking for gethostbyname - found

-- Looking for connect

-- Looking for connect - found

-- Looking for remove

-- Looking for remove - found

-- Looking for shmat

-- Looking for shmat - found

-- Found X11: /usr/lib/i386-linux-gnu/libX11.so

-- Found OpenGL: /usr/lib/i386-linux-gnu/libGL.so

CMake Error at cmake/Modules/FindPackageHandleStandardArgs.cmake:138
(message):

  Could NOT find FFTW (missing: FFTW_LIBRARIES FFTW_INCLUDES)

Call Stack (most recent call first):

  cmake/Modules/FindPackageHandleStandardArgs.cmake:374
(_FPHSA_FAILURE_MESSAGE)

  cmake/Modules/FindFFTW.cmake:20 (find_package_handle_standard_args)

  CMakeLists.txt:95 (find_package)

-- Configuring incomplete, errors occurred!

See also "/home/bob/CubicSDR/build/CMakeFiles/CMakeOutput.log".

*bob@bob-Latitude-E6400 ~/CubicSDR/build*

*bob@bob-Latitude-E6400 ~/CubicSDR/build *$ make

make: *** No targets specified and no makefile found.  Stop.

*bob@bob-Latitude-E6400 ~/CubicSDR/build $*

This build I started from my home directory but my previous attempt I would
have started from whatever directory I ended up in following the previous
steps, probably like Build and install SoapySDR and Build and install
liquid-dsp which went without problem.
I would have ended at  ~/soapySDR/build

vk2byf, 2015-11-30:
Sorry Charles I think my email was sent before I was finished editing. from
what I can see cmake failed because missing: FFTW_LIBRARIES FFTW_INCLUDES
 and then some more  which may as well be Chinese for me as I'm a raw
beginner way over my head but willing to learn.

Chanks for you help Charles, regards Bob

cjcliffe, 2015-11-30:
@vk2byf thanks -- I just noticed that the FFTW and wxWidgets dependency instructions had been misplaced for linux, I've added instructions back to the Wiki for Debian as follows:

```
sudo apt-get install libfftw3-dev libwxgtk3.0-dev
```

Try running that and let me know if it helps.

vk2byf, 2015-12-01:
Hi Charles
Getting closer.

I installed as per your reply above

sudo apt-get install libfftw3-dev
sudo apt-get install libwxgtk3.0-dev

The updated instructions on
https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux

include another package under Base Dependencies: (debian)

sudo apt-get install libfftw3-dev libwxgtk3.0-dev libwxgtk3.0-dbg

The error for the libwxgtk3.0-dbg

vk2byf, 2015-12-01:
Hi Charles
Getting closer.

I installed as per your reply above

sudo apt-get install libfftw3-dev
sudo apt-get install libwxgtk3.0-dev

The updated instructions on
https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux

include another package under Base Dependencies: (debian)

sudo apt-get install libfftw3-dev libwxgtk3.0-dev libwxgtk3.0-dbg

The error when trying to install libwxgtk3.0-dbg was could not find

Now I get as far as cd x64. the "make" took quite a while to complete.

There is no x64 directory but I found a x86 directory so I went there
instead but it failed on ./CubicSDR

I'm running Linux Mint installed as a 32 bit system.

git clone https://github.com/cjcliffe/CubicSDR.git

cd CubicSDR                 ok
mkdir build                 ok
cd build                    ok
cmake ../                   ok
make                        ok
cd x64                      bash: cd: x64 No such file or directory
I'm on ~/CubicSDR/build $
./CubicSDR

Thanks again for your efforts, Regards Bob

cjcliffe, 2015-12-01:
@vk2byf that actually seems about right; 32-bit systems will build to x86/ sub-folder (I'll have to make a note of that).

Is there a CubicSDR binary in x86/?  Can you run it and if not what is the output?

You might also try for CubicSDR:

```
sudo make install
```
And then you can just run 'CubicSDR' from anywhere. 

Removing it can be done with:
```
sudo make uninstall
```

cjcliffe, 2015-12-01:
Instructions have been updated and made a bit more clear with relevant info from this issue.

vk2byf, 2015-12-01:
Hi Charles

In ~/CubicSDR/build/x86/ there are only files like CubicSDR.ico, some 
vera_sans_mono18.png and similar, also some vera_sans_mono18.fnt files.

Nothing to run

CubicSDR won't run from anywhere, always returns CubicSDR command not found.

ran sudo make install from ~/CubicSDR/build it did 1% then fatal error; 
pulse/error.h: No such file or directory #include <pulse/error.h>
compilation terminated.

Thanks again, regards Bob

cjcliffe, 2015-12-01:
@vk2byf whoops -- almost there; looks like another dependency I forgot this time :)

I've added to wiki already, but run:
```
sudo apt-get install libpulse-dev
```

Which will install the pulse audio dependency.

cjcliffe, 2015-12-08:
CubicSDR will now notify when SoapySDR support modules are missing :)

[image]

closed 2015-12-08 by cjcliffe