cjcliffe/CubicSDR #191: Can't compile on ubuntu 15.04 armhf

gabika1941, 2015-11-17:
I've tried to install this CubicSDR on my armhf board because this seemed to be the simplest to install. Well, everything went fine until I've started to compile the CubicSDR itself. make says this: 
CMakeFiles/CubicSDR.dir/build.make:54: recipe for target 'CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o' failed
make[2]: *** [CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/CubicSDR.dir/all' failed
make[1]: *** [CMakeFiles/CubicSDR.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2

What should I do now, and what did I do wrong? Please help me

cjcliffe, 2015-11-17:
@gabika1941 Is there anything more to the cmake error output or log file you can post?  Not quite enough there to understand the issue.   Are all the dependencies (wxwidgets 3.0, soapysdr, fftw, liquid-dsp) already installed?

gabika1941, 2015-11-17:
In file included from /home/odroid/CubicSDR/src/CubicSDR.cpp:14:0:
/home/odroid/CubicSDR/src/CubicSDR.h: At global scope:
/home/odroid/CubicSDR/src/CubicSDR.h:180:1: warning: missing initializer for member ‘wxCmdLineEntryDesc::type’ [-Wmissing-field-initializers]
 };
 ^
/home/odroid/CubicSDR/src/CubicSDR.h:180:1: warning: missing initializer for member ‘wxCmdLineEntryDesc::flags’ [-Wmissing-field-initializers]
/home/odroid/CubicSDR/src/CubicSDR.h:180:1: warning: missing initializer for member ‘wxCmdLineEntryDesc::type’ [-Wmissing-field-initializers]
/home/odroid/CubicSDR/src/CubicSDR.h:180:1: warning: missing initializer for member ‘wxCmdLineEntryDesc::flags’ [-Wmissing-field-initializers]
/home/odroid/CubicSDR/src/CubicSDR.h:180:1: warning: missing initializer for member ‘wxCmdLineEntryDesc::shortName’ [-Wmissing-field-initializers]
/home/odroid/CubicSDR/src/CubicSDR.h:180:1: warning: missing initializer for member ‘wxCmdLineEntryDesc::longName’ [-Wmissing-field-initializers]
/home/odroid/CubicSDR/src/CubicSDR.h:180:1: warning: missing initializer for member ‘wxCmdLineEntryDesc::description’ [-Wmissing-field-initializers]
/home/odroid/CubicSDR/src/CubicSDR.h:180:1: warning: missing initializer for member ‘wxCmdLineEntryDesc::type’ [-Wmissing-field-initializers]
/home/odroid/CubicSDR/src/CubicSDR.h:180:1: warning: missing initializer for member ‘wxCmdLineEntryDesc::flags’ [-Wmissing-field-initializers]
/home/odroid/CubicSDR/src/CubicSDR.cpp: In function ‘long long int strToFrequency(std::string)’:
/home/odroid/CubicSDR/src/CubicSDR.cpp:97:16: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (numLen == std::string::npos) {
                ^
CMakeFiles/CubicSDR.dir/build.make:54: recipe for target 'CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o' failed
make[2]: *** [CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/CubicSDR.dir/all' failed
make[1]: *** [CMakeFiles/CubicSDR.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2

These are the last lines of the make's output.

mbott001, 2015-11-17:
Exactly the same error I ran into when I tried to install on Ubuntu 
15.10. I had made sure that all the dependencies that I knew of were 
met. The 14.04 install was without issue.

gabika1941, 2015-11-17:
Every dependency (wxwidgets 3.0, soapysdr, fftw, liquid-dsp) is installed, by the way

cjcliffe, 2015-11-18:
Hmm thanks but not getting much more from those, I'll give it a try here on ubuntu 15.10 soon; I'll chalk this up as a bug for the moment.

vsboost, 2015-11-18:
I compiled on Ubuntu 15.10 few days ago and cubicsdr works

vsboost, 2015-11-18:
If i remember correctly i had some issue building wxwidgets but i just used the line you have in the wiki under the osx section removed the mac related stuff and it worked.  Something like..

./configure --with-opengl --disable-shared --enable-monolithic --with-libjpeg --with-libtiff --with-libpng --with-zlib  --disable-sdltest --enable-unicode --enable-display --enable-propgrid --disable-webkit --disable-webview --disable-webviewwebkit

and used sudo make.

Sorry bit vague but it was a busy day..

gabika1941, 2015-11-18:
I've installed wxwidgets as a normal ubuntu/debian package from the ubuntu repo

cjcliffe, 2015-11-18:
In particular the '--enable-propgrid' option is new to the wiki and required for the devices dialog so it's easy to overlook if you had a version built and installed previously.

pcotret, 2015-11-28:
[test.txt](https://github.com/cjcliffe/CubicSDR/files/46410/test.txt)

I'm replying here as it seems I have the same error. In the file above, you'll find a log of what happened when building CubicSDR. I tried it on an armhf architecture with Debian Jessie.

Hope it can help.

Pascal.

cjcliffe, 2015-11-29:
@pcotret interesting bit in your log, no actual compile errors but:
```
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
```

Looks like an actual compiler issue; but I can definitely clean up some of those warnings in case they're related to it's ultimate failure.

guruofquality, 2015-11-29:
> c++: internal compiler error: Killed (program cc1plus)

@pcotret You could be running out of system memory on a particularly large source file. Adding a swapfile can help if that's the case.

cjcliffe, 2015-11-29:
@guruofquality that makes sense, I'd place my bets on '347K - RtAudio.cpp' even some of my editors warn me about it being too big :-)

pcotret, 2015-11-29:
@guruofquality Yes, adding a swapfile solved my problem. CubicSDR works now, thanks ;)

cjcliffe, 2015-12-02:
Glad to see it wasn't something more extensive :)   Closing this one for now, thanks!

closed 2015-12-02 by cjcliffe

rgood123, 2016-02-13:
Iam trying to install on debian jessie armhf cant get pass "make comand"
ERROR:
root@/liquid-dsp# make
gcc -I . -I include -Wall -fPIC -g -O2 -ffast-math -mcpu=cortex-a8 -mfloat-abi=softfp -mfpu=neon -c -o src/libliquid.o src/libliquid.c
In file included from /usr/include/features.h:398:0,
from /usr/include/complex.h:25,
from include/liquid.h:78,
from src/libliquid.c:27:
/usr/include/arm-linux-gnueabihf/gnu/stubs.h:7:29: fatal error: gnu/stubs-soft.h: No such file or directory
# include
^
compilation terminated.
: recipe for target 'src/libliquid.o' failed
make: *** [src/libliquid.o] Error 1
Thanks for your help

pcotret, 2016-02-13:
You just need to change a flag when you call GCC :
-mfloat-abi=softfp
must be replaced by -mfloat-abi=hard

cjcliffe, 2016-02-13:
@rgood123 @pcotret In specific the configure.ac fix by @andresv can be found here: https://github.com/jgaeddert/liquid-dsp/issues/7