cjcliffe/CubicSDR #195: New demodulator update broke squelch

Dantali0n, 2015-11-21:
Hi,

The new demodulator patch of last night broke the squelch on my ubuntu 14.04 lts machine,
Haven't tested it on windows yet but I expect the same results.

No matter where I set the threshold I always hear audio.

cjcliffe, 2015-11-21:
Aye, just forgot to finish this up -- going to make the ModemAnalog and ModemDigital return their own signal level and range instead of trying to generalize the squelch as it was (which was rather imbalanced between modem types)

I think the 'auto' might be broken for digital at the moment as well; but I've got a box of 900Mhz digital usb/serial modems here I'm going to set up in known conditions and see what I can get going.

cjcliffe, 2015-11-21:
Squelch should be fixed up now; new code might still need some tweaks but is functional again.

closed 2015-11-21 by cjcliffe