cjcliffe/CubicSDR #196: activateStream failed - sdrPlay

jdeitch, 2015-11-21:
Debian Wheezy, latest git pull.

If I start CubicSDR by selecting the sdrPlay I get this error:

[CubicLOG.txt](https://github.com/cjcliffe/CubicSDR/files/40944/CubicLOG.txt)

If I plug in a RTL dongle, start CubicSDR by selecting the RTL first then switching to the sdrPlay all is well.  Program uses the sdrPlay fine.

I am stumped.  Any help please.

Jim
JG1KGS

cjcliffe, 2015-11-22:
Hmm, that is strange I can't see why loading one before the other would matter unless there's some device interfering on the usb bus or I've introduced a race of some sort..

- Does it work the first time if you run as root?
- If you unplug any other devices you can from USB is there any difference?

I've got a few things on my list to test on Linux; I updated to Debian wheezy last week so I'll take another look myself soon as well.

jdeitch, 2015-11-23:
No.  
The only other things attached to the USB are the keyboard and mouse.  

One thing I did notice, the Mirics API installer installs a different libsdrPlaySupport.so file than the one I build from Pothos.  If I use the one from Mirics I get the above behavior.  If I use the one I build no matter what order they are loaded in trying to load the SDRPlay device fails.  They only differ by a couple hundred bytes so not exactly sure what it is.

Jim
JG1KGS

jdeitch, 2015-11-23:
Charles,
Another thing I notuced, in the UI the ppm choice for the SDRPlay is no longer available.  Was this by choice?

Jim

cjcliffe, 2015-11-23:
@jdeitch just wondering --  have you built and installed the latest versions from https://github.com/pothosware/SoapySDR and https://github.com/pothosware/SoapySDRPlay ? (and then rebuilt CubicSDR of course)

There are some minor changes between release versions and commits at the moment, I'd be interested in knowing if the master branches work -- and if not can you step back through the commits with ```git checkout HEAD~``` each time and build/install to see if there's somewhere it started happening?

As for the PPM display that will be gone until we actually add support for it in SoapySDRPlay.  Up until now all of the menu items have been hard-coded for RTL-SDR and adapted where needed (and some had no effect) -- now they can be dynamically created and populated just by querying the Soapy module which returns information about the various parameters and their allowed types/values for any device.

jdeitch, 2015-11-23:
Latest pull was 2300 UTC Sunday (0800 Monday in Japan).  

Sorry but I am a total novice with git.  After I do the git checkout HEAD~ what do I do?

Jim

cjcliffe, 2015-11-23:
@jdeitch Just re-make/install via ```make``` and ```sudo make install``` for Soapy projects, ```make install``` and ```make uninstall``` are available for CubicSDR as well now.   Make sure to rebuild any Soapy stuff before CubicSDR.   

doing checkout HEAD~ should just step one commit back in the history; so each time you do it you're moving back through the source.

I think to reset back to current it's ```git checkout origin/HEAD```.

jdeitch, 2015-11-24:
@cjcliffe  I found the problem!  Your coding is good, excellent to a novice like me actually.  What it was is I had 3, yes 3, different versions of libusb1.0 on my system.I was getting the include file from one version, the library from another.  The whole system was confused.  I rebuilt everything after I cleaned up everything and it worked.  Now to try and help you find that memory leak...

v/r,
Jim
JG1KGS

cjcliffe, 2015-11-24:
Good to hear that fixes it -- still strange though; RTL-SDR must initialize libusb in a way that SDRPlay was able to use it afterwards..   

Closing this one, cheers.

closed 2015-11-24 by cjcliffe