cjcliffe/CubicSDR #197: Frecuency drift?

Toontje, 2015-11-22:
I am listening to one station on the 20m SSB contest using the SDRPlay with CubicSDR connected to my TS-590. I noticed that i constantly need to adjust the frequency to get a clear audio from CubicSDR. When i listen to the same station using the TS-590 i don’t need to adjust anything. I switched from the SDRPlay to an RTL adapter and i notice the same. Frequency drift in CubicSDR?? If you look good you can see the drift on the below image (i deliberately put the filter edges next to the signal to show the drift.).
[image]

Toontje, 2015-11-22:
Double image just above the correct frequency.
[image]

cjcliffe, 2015-11-22:
Frequency drift I see fairly frequently with a RTL dongle while it's warming up but it's usually nice and constant -- I'm not familiar with the seemingly random drifts in the first few screenshots.

The last one I'm guessing you're right at the center frequency (and you won't see that in the upper-left spectrum); that's a side-effect of the visual DC removal it duplicates the visual data nearby over-top of the DC spike at center frequency.

Either way the input code for those visuals hasn't changed recently and any drift of that type would likely be occurring at the device / source transmission as far as I know.

Toontje, 2015-11-22:
So you are telling me that the Play as well as the RTL adapter have similar characteristics? I can adhere to the fact that these measurements were taken when both radios were still warming up. Now, compared to any other test i have done in the past these were both taken when the adapters were used as panadapters for the TS-590 where the Play gets its signal from the antenna RX circuit and the RTL adapter gets its signal from one of the two IFs in the radio. Maybe that added a variable to the mix. I have never seen a drift this much with any of the RTL adapters or with the Play. Anyway, higher priorities than this one. Closing the issue.

closed 2015-11-22 by Toontje

vsboost, 2015-11-22:
@Toontje Can you test with a different SDR software?

k9wkj, 2015-11-22:
looks like the IF tap is loading down the circuit a bit
thats the only way there can be a shift
it may need a resistor to drop the current a bit
but then again  maybe im nutzzz

k9wkj, 2015-11-22:
added info
or a bit lighter coupling if thats how its done (have a worse match on the
impedance to drop the current down)