cjcliffe/CubicSDR #198: LNAT

Toontje, 2015-11-22:
What does the LNAT control do? I don't see any difference wether it is set to 0 or to 101 (not 100, 101... ;-))

cjcliffe, 2015-11-22:
It's not a direct gain value per-se; LNAT = "LNA Threshold" for SoapySDRPlay; it's the "LNA Trigger" level -- if you set it below the current gain it'll activate the LNA; above and it'll disable it.

Toontje, 2015-11-22:
Did you notice it goes to 101 instead of to 100? Is that intentional?

SDRplay, 2015-11-22:
guys I wouldn't put too much effort into this - the scale is wrong - the LNA gain threshold only goes from 24-59. We are rewriting the SoapySDRPlay module to work in the same way as the EXTIO functions do. Hopefully we'll have something ready in the next few days to test.

closed 2015-12-01 by cjcliffe