cjcliffe/CubicSDR #200: Which to build?

vsboost, 2015-11-27:
Dumb question forgive me im not an expert at github.

I notice you have a separate branch for digital_experiments, how does that work exactly? do i need to build that separately from the master.

Sorry

cjcliffe, 2015-11-27:
Ahoy, it's pretty straightforward:

```
$ git fetch
.. should pull in some branches ..
$ git checkout digital_experiments
```

You can get back to master with:
```
$ git checkout master
```

Cheers!

vsboost, 2015-11-27:
Thank Charles,

Was worried that by building the master i wasn't getting the stuff you have done in the digital_experiments branch.

cjcliffe, 2015-11-27:
No problem; let me know if you spot any issues; there's a bit of cleanup to do and I'm working on adding something simple to display the digital data output.

vsboost, 2015-11-27:
Will do.

closed 2015-11-27 by vsboost