cjcliffe/CubicSDR #201: Input bandwidth and SampleRate menu

frspin, 2015-11-27:
Why menu named "Input bandwidth" enumerate getSampleRates() returned values?
In SDRPlay samplerate as returned by listSamplaRates(9 is correct but Input bandwidth is returned by listBandwidths() function.

Default value is 2.024 Mhz, correct as Samplerate value but incorrect as Input Bandwidth (1.536 Mhz)

Regards
Franco Spinelli
IW2DHW

cjcliffe, 2015-11-27:
There's a mix of old a new naming conventions at the moment, with RTL-SDR I called it "input bandwidth" but for SoapySDR that's the Sample Rate and Bandwidth is IF bandwidth.  

I'll make some corrections to the terms; they haven't been updated in a long time :-)

frspin, 2015-11-27:
Il 27/11/2015 22:53, Charles J. Cliffe ha scritto:
> There's a mix of old a new naming conventions at the moment, with
> RTL-SDR I called it "input bandwidth" but for SoapySDR that's the Sample
> Rate and Bandwidth is IF bandwidth.
> 
> I'll make some corrections to the terms; they haven't been updated in a
> long time :-)
> 
> —
Ok, so menu named "Input bandwidth" is populated with samplerate.
But I can't see an "IF bandwidth" menu for setting it  to 1.536 Mhz
And decoding is changing if I change menu value from default 2.048 Mhz
to 1.536 Mhz.
1.536 Mhz is a valid bandwidth for SDRPlay, 2.024 Mhz is invalid.

Regards
Franco Spinelli
IW2DHW

cjcliffe, 2015-11-28:
Just as a note the latest SoapySDRPlay automatically enables Low-IF mode with down-sampling for 2.048Mhz, when it's running at 2.048Mhz (and some lower modes) it's actually running at 8Mhz internally and down-sampling to 2.048Mhz so the 1.536Mhz bandwidth does not apply.

cjcliffe, 2015-12-08:
Naming has been updated in master branch, thanks!

closed 2015-12-08 by cjcliffe

frspin, 2015-12-08:
Il 08/12/2015 03:03, Charles J. Cliffe ha scritto:
> Naming has been updated in master branch, thanks!
> 
> —

OK
Now "Sample Rate" menu' correctly display available sample rate.

How can I modify input bandwidth? A reduced input bandwidth can reduce
CPU load or show, in HF, only full OM band (variable from 0.1 MHz to 1.3
MHz)

Regards

Franco Spinelli
IW2DHW