cjcliffe/CubicSDR #202: Mac OS X El Capitan strange issues

red-green, 2015-11-28:
I have been using CubicSDR for some time on my Mac, using OS X Yosemite. Recently, after updating to El Capitan, I discovered that CubicSDR isn't working properly. Before, the waterfall display had no detectable lag, but now the waterfall is almost useless because of lag. The audio demodulation seems to be unaffected.

cjcliffe, 2015-11-28:
Just wondering what version is this for? 0.1.4 has some serious issues with waterfall lag in 10.11 and the mouse issue you described that are fixed in 0.1.6+ releases here on GitHub.   

Latest SoapySDR release should be stable on El Capitan as I've been testing with it: https://github.com/cjcliffe/CubicSDR/releases/tag/0.1.16-soapysdr

dhoelzer, 2015-11-28:
I haven’t messed with Cubic on Capitan, but I can tell you that GnuRadio is less than thrilled… It’s actually the RTL and Osmocom sources that seem to be unhappy.  Starting a flowgraph causes a 10-20 second beachball while it figures out the SDR device.  Under Yosemite there is zero delay.

fyngyrz, 2015-11-28:
cjcliffe - I sent you an email about rights and another issue a day ago... I see you're active WRT the project, but I have no reply to the email. Is the email on your site a poor or dead channel? Is there another means for non-public discussion you prefer?

Thank you

--Ben
fyngyrz aht gmail daht com

red-green, 2015-11-28:
I've noticed that too... Rtl_fm is much slower than it used to be

guruofquality, 2015-11-28:
> I haven’t messed with Cubic on Capitan, but I can tell you that GnuRadio is less than thrilled… It’s actually the RTL and Osmocom sources that seem to be unhappy. Starting a flowgraph causes a 10-20 second beachball while it figures out the SDR device.

Sounds familiar. Recently, a user mentioned the Osmocom source being slow issue. The workaround was to specify the specific device driver in the device arguments parameter. Example "hackrf=0": https://github.com/pothosware/PothosSDR/wiki/GNURadio#grosmosdr-blocks

That probably doesn't explain rtl_fm, but I think the right combination of dependencies for osmocom blocks might cause this. Which might explain one system vs another showing the issue.

dhoelzer, 2015-11-28:
Oo, that’s a great tip.  I’ll give that a try when I land.

dhoelzer, 2015-11-29:
This definitely improves the situation!  Thank you!

cjcliffe, 2015-12-05:
I think this is solved; @red-green 's issues are definitely from earlier builds plus bonus GnuRadio tips :)

closed 2015-12-05 by cjcliffe