cjcliffe/CubicSDR #203: Looking for a remote soapysdr server to "Add Remote" for testing within CubicSDR.

vielmetti, 2015-11-29:
I'm running 0.1.16 alpha on MacOS 10.9.5 using the new SoapySDR support to tune a local NooElec R820T2 NESDR Mini 2 local RTL-SDR receiver. It seems to work just fine (yay) though I haven't done that much testing.

I'm intrigued by the File > Add Devices > Add Remote option, but I wasn't able on first search to find any remote public servers to connect to for testing, if any of those even exist. 

I do know that there are a lot of public SDR servers indexed at http://websdr.org so the thought that someone would put some bandwidth and radio(s) available for use is not unheard of.

Thanks for any help for a brief or extended test if you have a server available. I can connect from both bandwidth-challenged and bandwidth-plenty locations if it's useful for testing.

cjcliffe, 2015-11-29:
Right now you'll need to run a SoapyRemote server http://github.com/pothosware/SoapyRemote and compile the SoapyRTLSDR module on that same device.  Eventually I'll include a SoapySDRServer runtime with CubicSDR so you can do it more easily.

If the device is on the local network it'll auto-discover and show under the "Local Net" section.  Otherwise you'll have to add the device by IP with "Add Remote".

RFSpace is technically supported through SoapyOsmo module (OSX Builds) but I don't think there's currently a way to add it as a remote properly.

Toontje, 2015-11-30:
I was thinking about the same and created a page to register publicly available receivers to be used with SoapyRemote here: https://github.com/cjcliffe/CubicSDR/wiki/Publicly-available-remote-receivers

cjcliffe, 2015-11-30:
@Toontje I think that's a great idea; I should be able to put a Pi or something online here

Toontje, 2015-12-01:
Thanks Charles. Let's see if other people also put their receivers online. I for sure will put my RTL adapter online when i get home (and have time to do it).
Please make a link to this page from the main page of your wiki so people know that it exists.