cjcliffe/CubicSDR #206: Implement remaining digital output streams

cjcliffe, 2015-12-02:
Modem outStream implementations so far:
  - [ ] APSK
  - [ ] ASK
  - [ ] BPSK
  - [ ] CPFSK (todo)
  - [ ] DPSK
  - [x] FSK (ENABLE_LIQUID_EXPERIMENTAL)
  - [x] GMSK
  - [ ] OOK
  - [ ] PSK
  - [ ] QAM
  - [ ] QPSK
  - [ ] SQAM
  - [ ] V29

Toontje, 2016-02-07:
I know i was one of the people initially asking for this, but thinking about it, should you continue with this feature or should you create an option to redirect data streams to external programs like FLDigi? There are so many good decoding programs out there so a simple data redirect could be an option to get some work off your hands leveraging already existing programs.
Unless you are panning to create a protocol detection algorithm like in Hoka Code, Wavecom, Krypto500, etc.

vsboost, 2016-02-07:
I would still like this feature.

Toontje, 2016-02-07:
But for which (read: how many) modulations? There are hundreds and everybody would like a different one. That's why programs like FLDigi are so popular. They support those hundreds.

cjcliffe, 2016-02-07:
@Toontje generally it's going to be a front-end to do other things is why I haven't expanded too much on the current implementation.  I'll need to verify that everything is going to work as expected as a plug-ins before I fully define how it works and solidify the output routines.

You'll be able to redirect it to block output, network or run a custom LUA script on it; any new plug-ins that need digital decoding can use the existing modems as a base to keep the decoding/encoding process plug-in code simple and up-to-date.

Additionally there's going to be visual outputs such as a pixel buffer that scripts can use to process visual data which I haven't defined yet.  There's some interesting use cases for that including Mike R.'s passive radar display code which I think will be popular.

So definitely still on the table but I've sort of put it on hold until after the 0.2.0 beta release and related issues are cleared up.

vsboost, 2016-02-07:
Personally for me the less I have to rely on 3rd party apps the better, plus I am interested in the binary data generated from gmsk / fsk / ask / ook

Dantali0n, 2016-02-08:
Me to one of the key features that differentiates CubicSdr is that you would have access to raw binary, add plugin support onto that and you could even design your own protocol!

cyber5tar86, 2016-03-21:
Hi!
I've some suggestions for the digital lab GUI development: on internet I found the Sora Microsoft Project related to software radio:

  - http://research.microsoft.com/en-us/projects/sora/
  - https://github.com/Microsoft/Sora

The PDF manual (http://research.microsoft.com/pubs/160799/SoraManual-v17.pdf) on page 90 shows the GUI, from which we could take hints to develop the digital lab GUI

tygamvrelis, 2019-10-26:
Just came across this while looking for a way to demodulate QAM—is there currently a way to do this within Cubic, or redirect data streams?

Dantali0n, 2019-10-28:
> Just came across this while looking for a way to demodulate QAM—is there currently a way to do this within Cubic, or redirect data streams?

Yes you can but I don't think it will satisfy your needs. If you use the digital lab the output stream will be placed into a text field and there is no real way to redirect this stream. Naturally this really slows down Cubic. 

Nothing is done to check if the demodulation is successful so all data is streamed into the text field all the time regardless of the input signal. In addition, nothing is done to compensate for any slight offset from the center frequency so any slight offset up or down in frequency can result in invalid data.