cjcliffe/CubicSDR #208: Pen touch input

vsboost, 2015-12-03:
Hi Charles,
Not sure if its CubicSDR or something else but using Ubuntu on my surface pro CubicSDR wont allow me to select anywhere on the waterfall using the pen, it works on other parts of the program like changing frequencies / selecting menu items or modulation types, i have to use my finger to make a selection on the waterfall.

Hope that makes sense.

Cheers

cjcliffe, 2015-12-15:
@vsboost I'm guessing it's almost impossible to lift the pen without moving a few pixels -- CubicSDR's tolerance for a drag vs. a tap is very small -- I may have to tweak it a bit to allow a bit of movement to still register as a click instead of drag..

vsboost, 2015-12-15:
You are correct, if i am ultra stable with the pen when i click it works as it should.