cjcliffe/CubicSDR #209: Small UI enhancement requests

kingsqueak, 2015-12-03:
Some small UI enhancements, that I tried to do and submit, but I'm a frighteningly bad coder and failed :-)

1) a one pixel reticle/line that would indicate center of tune while hovering on the demodulator both in spectrum and in the FFT box - use case is to have a defined indicator for doing PPM adjustments as an example. The line should show up through the frequency spectrum to center on a carrier I actually like that there isn't one there all the time, but on hover would be nice to highlight and show center tune as a clear line.

2) When hovering over the slider for the gain and other items, a floating popup to indicate the value at the hovered point would be useful. - use case - With gain the difference between say 33 and 34 can be a threshold value and it's tricky to try and click to exactly 33.

3) One for the farrrr back burner, but support for theme files for color schemes would be cool to make it easier to skin it if a user wanted and to easily share the overall theme.

Thanks for doing this....running it on Ubuntu 14.04 with the SDRPlay and it works great.

cjcliffe, 2015-12-03:
Thanks for the suggestions! All sounds good; adding it to the queue for enhancements.

cjcliffe, 2016-03-15:
Moving the last part of this to after the 0.2.0 milestone; need to unify some UI to make the floating value part properly.