cjcliffe/CubicSDR #210: Audio output changes back to default device

vsboost, 2015-12-04:
Hi Charles,

Noticed that if i select a different audio output device, in my case (Cable Input VB Audio Cable) under windows, if i change center frequency and click on part of the waterfall the audio output switches back to the default device.

This does not happen if i change frequency using the frequency box only happens if changing frequency using the center frequency box and clicking part of the waterfall.

cjcliffe, 2015-12-04:
Hmm, that menu is really just for setting the current modem audio output for piping; at the moment there's no "default audio device" selection -- I think that's what's needed here.

vsboost, 2015-12-05:
Hi Charles, this one is through windows,

[image]

When i select VB audio cable everything is fine until i change frequency using the centre frequency box, once i select part of the waterfall after i have changed freq the Audio Output goes back to default device which then pipes the audio to the speakers.

ab1jx, 2019-09-09:
When I had my CubicSDR compiled with PulseAudio I could choose audio devices like that (but there was only one).  I recompiled to use Alsa instead of PulseAudio and that disappeared from the menu.  But really I have 3 Alsa subdevices for HDMI, analog, spdif all in "card" 0 on this Odroid N2.  I want card 0, device 1 (analog).  That's the default in my /etc/asoundrc but I've gotten no sound from CubicSDR yet.  It lists them all on startup correctly, I want to default to Audio Device #1 hw:AML-AUGESOUND,1 (in this case).  I get sound from other programs OK.