cjcliffe/CubicSDR #211: High CPU usage

vsboost, 2015-12-04:
Hi Charles,

Not sure if its normal behaviour but i'm getting pretty high CPU usage demodulating FM

Tasks: 254 total,   2 running, 252 sleeping,   0 stopped,   0 zombie
%Cpu(s): 22.9 us,  7.1 sy,  0.0 ni, 68.9 id,  1.2 wa,  0.0 hi,  0.0 si,  0.0 st
KiB Mem:   8093964 total,  1466728 used,  6627236 free,    64912 buffers
KiB Swap:  8307708 total,        0 used,  8307708 free.   727464 cached Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND    
 2435 me       20   0 1987604 174052  37216 S  92.7  2.2   0:39.28 CubicSDR   

Also when attempting to demodulate GMSK after about 10sec everything slows to a crawl and eventually the waterfall stops moving.

Tasks: 249 total,   2 running, 247 sleeping,   0 stopped,   0 zombie
%Cpu(s): 44.0 us,  1.9 sy,  0.0 ni, 53.9 id,  0.3 wa,  0.0 hi,  0.0 si,  0.0 st
KiB Mem:   8093964 total,  1827076 used,  6266888 free,    71100 buffers
KiB Swap:  8307708 total,        0 used,  8307708 free.   879936 cached Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND    
 2576 me       20   0 2000304 186344  37600 R 169.4  2.3   0:50.80 CubicSDR   

Machine is a Surface Pro2 / i5 / 8gb ram

cjcliffe, 2015-12-04:
Interesting, I'll have to do some more windows testing -- there may be some thread lock / race issues related to the new console stuff.  A for the higher CPU usage I noticed that Windows frame rate is now getting a bit excessive again since the main thread has been freed up significantly -- I need to do some more work on throttling it back to 30fps as I'm not sure the frame interval is working correctly in Windows 10 now.

vsboost, 2015-12-05:
Hi Charles, I should have said this was under Linux.

cjcliffe, 2015-12-05:
Ok, that changes things a bit :)  I'll get to doing some performance testing in Linux and see what I can find.  There's likely some new issue with the amount of code that's changed recently.

closed 2016-08-17 by cjcliffe