cjcliffe/CubicSDR #214: Noise Reduction/Filter

cjcliffe, 2015-12-08:
From #179 
- [ ] Basic noise reduction toggle
- [ ] Visual notch filter
- [ ] Advanced filters, type selection and visual controls

Toontje, 2016-02-11:
Looking forward for you to get this issue closed. ;-)

TonMachielsen, 2016-07-26:
Hey Charles! I'm back!! Still working on this issue? It's from last year.... ;-)

cjcliffe, 2016-07-26:
@TonMachielsen good to see you back; new account?  I've been thinking about the UI for this for awhile -- I'm going to start filing the issues into releases once 0.2.0 stable is out and then we can start to go a bit nuts in experiment branches :)

TonMachielsen, 2016-07-27:
Ay, crap! I responded wit the wrong account. Yes, i have one work account and one personal account. I guess i'd need to merge them somehow. 
Yeah, i saw a lot of new issues. They are keeping you busy. ;-)