cjcliffe/CubicSDR #218: v0.1.18-alpha crash OS X

effeerre75, 2015-12-09:
Error is "Segmentation fault: 11" if some modules are present in /usr/local/lib/SoapySDR/modules/
Works fine if the modules are removed.

dc1rdb, 2015-12-09:
Works fine here under 10.9.5 with the following modules (current build) present in /usr/local/lib/SoapySDR/modules/
libHackRFSupport.so
librtlsdrSupport.so
libsdrPlaySupport.so

cjcliffe, 2015-12-09:
Will investigate; likely an issue with an almost-compatible module with the same version attempting to load.

effeerre75, 2015-12-10:
Had some time to check the error is

SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/librtlsdrSupport.so)
  rtlsdr failed ABI check: Library ABI=0.4-2, Module ABI=0.3-0

Placing the modules that come inside the bundle in  /usr/local/lib/SoapySDR/modules/ 
CubicSDR start and works fine

Placing the module from /usr/local/lib/SoapySDR/modules/ in your bundle 
CubicSDR start but didn't works.

It think I had to make a clean install of my Soapy. If you want I'll wait for testing it ;)

cjcliffe, 2016-01-09:
Can't reproduce with anything but a compiled binary mismatch, this will improve with time as SoapySDR module versions stabilize so closing for now.

closed 2016-01-09 by cjcliffe