cjcliffe/CubicSDR #219: IF Bandwidth

frspin, 2015-12-09:
After recent improvements CubicSDR  now have all SDRPlay Extio control except for IF Bandwidth and Tuner DC Offset
IF Bandwidth is good for reducing CPU usage on small processor. And a bandwidth of 300 or 600 KHz is enough for many OM bands.

For audio DSP  there are other already opened issues.

A last improvement can be a band filter non only of variable width (as now) but also with an adjustable starting and ending frequency, for "isolate" a signal. This can be done also in audio DSP.

Regards
Franco Spinelli
IW2DHW

cjcliffe, 2015-12-09:
Just as a note, the latest SoapySDRPlay master branch utilizes IF Bandwidth and Low-IF modes automatically and can go as low as 222KHz Sample Rate.

@SDRPlay is working on a new version of SoapySDRPlay that I believe will be closer to the ExtIO plugin features and I hope to use the result to improve the SoapySDR settings interaction and re-create the SDRPlay GUI (with perhaps a bit more graphical features) via issue #194

frspin, 2015-12-10:
2015-12-10 0:18 GMT+01:00 Charles J. Cliffe <notifications@github.com>:

> Just as a note, the latest SoapySDRPlay master branch utilizes IF
> Bandwidth and Low-IF modes automatically and can go as low as 222KHz Sample
> Rate.
>

How IF Bandwidth  can be modified? In master branch (last update 29 days
ago) I can see a function setBandwidth. Can I reduce bandwidth reducing
SampleRate?

> @SDRPlay <https://github.com/SDRPlay> is working on a new version of
> SoapySDRPlay that I believe will be closer to the ExtIO plugin features and
> I hope to use the result to improve the SoapySDR [...]

It will be a very nice feature

Regards

Franco Spinelli
IW2DHW

frspin, 2015-12-10:
I have compared CubicSDR and ExtIO with HDSDR (running under Wine in Linux)
Both display a range depending from samplerate, as correct.

CubicSDR show a bandwidth according to range on display (and so to sample rate)

Using ExtIO  you can set a bandwidth and samplerate change for this, or you can set a samplerate and a bandwidth below samplerate. In this case on display you can see only a small portion with signal.

SoapySDRPlay have both samplerate and bandwidth value so what is needed is a menu with bandwidth values and correlation between bandwidth and samplerate values.

Hope this help

Franco Spinelli
IW2DHW

jazzkutya, 2015-12-14:
In CubicSDR the input bandwidth setting is actually sdrplay sample rate. SoapySDR sets IF filter bandwidth automatically according to selected sample rate.

cjcliffe, 2016-02-05:
This will be addressed with an updated SoapySDRPlay in the near future

closed 2016-02-05 by cjcliffe