cjcliffe/CubicSDR #222: Sessions

jdeitch, 2015-12-12:
When loading a Saved Session the bandwidth is not set and the demodulator does not go active.  Is there a hotkey to switch between demodulators?

cjcliffe, 2015-12-12:
@jdeitch I'll take a look at improving the scope of save/load properties in the session.  There's a plan to add keyboard shortcuts and some light arrows for navigating them in the UI.  You can track the progress here: #91

cjcliffe, 2016-01-10:
@jdeitch latest commit to master will now load/save bandwidth and activate single-modem save files properly.

jdeitch, 2016-01-11:
@cjcliffe Works like a charm.  Thank you!

jdeitch, 2016-01-11:
@cjcliffe Correction to above.  The demodulator does not change to the one specified in the session file.

cjcliffe, 2016-01-11:
@jdeitch right, I forgot to handle the multiple demodulator scenario; will patch that up

cjcliffe, 2016-01-12:
Ok, properly fixed now -- CubicSDR will actually save the active demodulator flag to the session file and it will be active when loaded :)

closed 2016-01-12 by cjcliffe

jdeitch, 2016-01-13:
@cjcliffe Working wonderfully!  Thank you!