cjcliffe/CubicSDR #223: Possibility to label devices

Toontje, 2015-12-12:
[image]
I would like to label the detected devices like:
Panadapter
RTL with standard antenna
SDRPlay connected to radio

cjcliffe, 2015-12-12:
This should be pretty simple, I think I can enable editing names in the tree via simple right-click menu without much trouble

cjcliffe, 2016-01-19:
@Toontje latest commit to master you can now edit device names :)

Toontje, 2016-01-19:
Great! I will try it out when i get back home on friday. Fantastic work!

cjcliffe, 2016-01-19:
@Toontje

cjcliffe, 2016-01-19:
@Toontje Switching devices while editing name or offset will now save changes to appropriate device -- good catch thanks :)

closed 2016-01-19 by cjcliffe