cjcliffe/CubicSDR #225: feature request: larger font size for main spectrum window frequency axis

ElectronicBattle, 2015-12-13:
Hello

The point size used in the main frequency spectrum display axis labelling (both the frequency and also the amplitude scale) - could it be enlarged please (perhaps via a user-selectable option)?

By way of contrast: when you add a demod, the font size associated with it, in the frequency spectrum window (and similarly in the zoomed-in window top left), is easily readable.

Thanks!

cjcliffe, 2015-12-13:
Makes sense; I might make it adjust it's size based on how large you drag the spectrum view now that it's resizable.

cjcliffe, 2016-01-14:
@ElectronicBattle latest commit to master will automatically adjust the spectrum font size; if it's too small or large just drag the spectrum divider and font will increase/decrease as appropriate.

closed 2016-01-14 by cjcliffe