cjcliffe/CubicSDR #226: openGL missing error

w9ran, 2015-12-13:
Hello - newbie trying to compile CubicSDR on a Raspberry Pi 2 platform (which I realize may be challenging).   However I get a cmake error for missing openGL and I did not see this among the dependencies:

CMake Error at /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:97 (MESSAGE):
  Could NOT find OpenGL (missing: OPENGL_gl_LIBRARY)
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:288 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-2.8/Modules/FindOpenGL.cmake:153 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:103 (find_package)

-- Configuring incomplete, errors occurred!

--

Any suggestions as to how to remedy this would be appreciated.   Thanks.

Bob W9RAN

cjcliffe, 2015-12-13:
@w9ran ahoy, not quite sure how to resolve that dependency on Raspberry Pi 2 at the moment but I should have one available soon and I'm planning to see what I can do to get it working.

mbott001, 2015-12-26:
I've encountered this with Ubuntu 14.04.3 LTS.
mbott@mlaptop:~/CubicSDR/build$ cmake ../ -DCMAKE_BUILD_TYPE=Release
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
64 bit compiler detected
CMake Error at /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:108 (message):
  Could NOT find OpenGL (missing: OPENGL_gl_LIBRARY OPENGL_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:315 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-2.8/Modules/FindOpenGL.cmake:161 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:103 (find_package)

cjcliffe, 2015-12-26:
I believe this will install the appropriate dependencies on systems that don't already have OpenGL headers / graphics driver installed, untested here at the moment as I'm using nVidia drivers on my linux installation(s) and they all apparently have the headers already:

```
sudo apt-get install freeglut3 freeglut3-dev
```

Let me know if that helps and I'll add it to the instructions

mbott001, 2015-12-27:
I'll do my best to try your recommendation later today.  Thanks!

mbott001, 2015-12-27:
Installing freeglut3 and freeglut3-dev was the solution for me.  Thanks again!

-- 
Mike

cjcliffe, 2015-12-27:
@mbott001 good to hear, I'll add that note to the Wiki

mbott001, 2015-12-28:
Glad I was able to help.
-- 
Mike

cjcliffe, 2015-12-28:
@w9ran any chance installing freeglut3 freeglut3-dev makes it work on the Raspberry Pi 2?

mbott001, 2015-12-28:
Just for grins, I followed the same instructions and installed CubicSDR on Linux Mint 17.3 Cinnamon. Worked like a charm.

w9ran, 2015-12-28:
Hi Charles,

Due to holiday travel I'm just getting able to try this today, but no luck:   http://pastebin.com/bmz4HrHj

This seems to point to another missing dependency...or could it be unable to find other components that I've previously installed (e.g. SoapySDR or liquid dsp)?    I have to admit it's been so long the trail is a bit cold, but after installing the freeglut3 that's what I got.

Thanks,

Bob W9RAN

cjcliffe, 2015-12-28:
@w9ran looks like it's just missing fftw3; but for good measure try:
```
sudo apt-get install libfftw3-dev libwxgtk3.0-dev libpulse-dev
```

cjcliffe, 2015-12-28:
@mbott001 good to hear, I haven't done Mint on my Linux install spree yet -- just Arch, Debian, OpenSUSE and Ubuntu so far, but will add it to the list.

w9ran, 2015-12-29:
On 12/28/2015 4:45 PM, Charles J. Cliffe wrote:
>
> @w9ran <https://github.com/w9ran> looks like it's just missing fftw3; 
> but for good measure try:
>
> |sudo apt-get install libfftw3-dev libwxgtk3.0-dev libpulse-dev |

Charles - that's a step closer, the other 2 packages install but 
libwxgtk3.0-dev seems to be in hiding.  I've tried several mirrors but 
apt can't find it (even though it's supposed to be in the Jessie and US 
debian repos).   And you were right, the build fails without 
wxWidgets.    I'll keep trying to find it.

Bob

cjcliffe, 2015-12-29:
@w9ran Running 
```
sudo apt-get update
```
and 
```
sudo apt-get dist-upgrade
```

might help.

w9ran, 2015-12-29:
On 12/28/2015 7:34 PM, Charles J. Cliffe wrote:
>
> @w9ran <https://github.com/w9ran> Running
>
> |sudo apt-get update |
>
> and
>
> |sudo apt-get dist-upgrade |
>
> might help.
>
>
Hi Charles,

Well it did sound like a good idea, but I guess no good idea goes 
unpunished ;-)   In the middle of the dist-upgrade it ran out of memory 
and I'm not back to square zero with a fresh install of raspbian 
jessie.    I may try to see if I can build CubicSDR on another linux 
machine before I re-attempt the Pi2.   Has anyone else done so with 
success or are these uncharted waters?

Thanks for the support.  I play with this stuff in order to learn and - 
I'm learning!

Bob

cjcliffe, 2015-12-29:
@w9ran whoops, might be a good idea to add a swap partition; it's likely that even compiling CubicSDR could run out of memory as well.

I should have a chance to try it here soon, cheers!

mbott001, 2015-12-29:
On 12/29/2015 12:50 PM, w9ran wrote:
> Well it did sound like a good idea, but I guess no good idea goes
> unpunished ;-) In the middle of the dist-upgrade it ran out of memory
> and I'm not back to square zero with a fresh install of [...]
I've built and run on both Ubuntu 14.04LTS and Linux Mint 17.3.

-- 
Mike

w9ran, 2015-12-29:
On 12/29/2015 12:12 PM, mbott001 wrote:
> I've built and run on both Ubuntu 14.04LTS and Linux Mint 17.3.

Good, I've got 14.04LTS on a laptop, I'll try it next.  Sorry I wasn't 
clear, but wondered if anyone had tried/succeed with installation on 
Raspberry Pi2, with Raspbian Jessie.

Bob

w9ran, 2015-12-29:
Hi Charles,

Tried building CubicSDR on a laptop with Xubuntu 14.04LTS.  The good 
news is, the build went perfectly, it only needed a couple of the 
dependencies which you'd provided earlier, and no errors.   The bad news 
is, it won't run.   The first time I did see the SDR windows in the 
background but a couple of foreground windows were frozen - one said 
something about SoapySDR modules and the other only displayed some 
unintelligible characters.   After that it refused to load any of the 
graphic screens with this error which appears related to threading:

www.pastebin.com/U1NFNkS4

I did install it as root, if that makes a difference.  When I try 
running as root I get the following error:

www.pastebin.com/YiFwVjW4

Any thoughts would be appreciated,  thanks,

Bob

k9wkj, 2015-12-29:
hey Bob !
i would be suprised if the pi could run Cubic (very suprised really)
ive tried a bunch of stuff with my pi and really i can only get it
to be spectrum server
that said, i think rtl_tcp is still broken badly (ive never gotten it to
work properly, even on my octacore machine it stutters unusably)
would sure be nice to use my Ranverter remotely !!

w9ran, 2015-12-30:
On 12/29/2015 3:30 PM, k9wkj wrote:
> that said, i think rtl_tcp is still broken badly (ive never gotten it to
> work properly, even on my octacore machine it stutters unusably)
> would sure be nice to use my Ranverter remotely !!
Not to derail onto this topic, but several changes now allow rtl-tcp to 
work quite well, including updates to the rtl-tcp plug-in for SDR# and a 
newer version of libusb that is part of Raspbian Jessie.    I think if 
you update your Pi and use the latest version of SDR# on the client PC, 
you'll be much more satisfied.   I can't speak to if or how well it 
works with any other SDR apps.

73, Bob W9RAN

cjcliffe, 2015-12-31:
I wouldn't dismiss the Pi2 being able to run CubicSDR yet;  it has already been reported to run on the similar "Banana Pi" successfully in the past -- and it's only a dual-core chip..

k9wkj, 2015-12-31:
the big trouble with the pi is that all the hardware hangs on the usb buss
so its shackled with the usb2 stutters

w9ran, 2016-01-01:
On 12/31/2015 12:36 PM, k9wkj wrote:
> the big trouble with the pi is that all the hardware hangs on the usb buss
> so its shackled with the usb2 stutters

That's a commonly-held perception, yet work I've been doing indicates 
that properly structured multithreaded code can produce pretty amazing 
results.  Here is the Pi 2 receiving ADSB data at 20 million samples per 
second via USB and sending the decoded data via ethernet to a PC, with 
sufficient resources left to display the resulting live Google map in 
the Pi's Epiphany browser:

https://twitter.com/w9ran/status/682613392629092352

If you examine the htop display you will see that even under this load, 
all four cores are running around 75-85% and none are saturated.   
That's pretty impressive performance for what many people consider to be 
a "toy".

73, Bob W9RAN

cjcliffe, 2016-01-09:
Latest version I've been able to compile and run on a Raspberry Pi2 (sort of..) with some tweaks to the Linux instructions.  OpenGL contexts are blank and don't appear to render -- might be a simple issue of correcting OpenGL initialization, but investigation is to come.

I have the Pi2 hardware here now (thanks to some donations on cubicsdr.com) and will be able to work on getting it running properly.

cjcliffe, 2016-01-12:
I've added everything learned here to the Linux build page; going to move this over to new Raspberry Pi2 discussion at issue #242 to avoid fragmenting.

closed 2016-01-12 by cjcliffe

w9ran, 2016-02-20:
On 12/31/2015 11:25 AM, Charles J. Cliffe wrote:
>
> I wouldn't dismiss the Pi2 being able to run CubicSDR yet; it has 
> already been reported to run on the similar "Banana Pi" successfully 
> in the past -- and it's only a dual-core chip..
>
>

Charles -

I've been away from this issue for a while and just wonder if you might 
have an update on the feasibility of running CubicSDR on the Raspberry 
Pi 2?

 From the latest post on issue #242 it appears that an improvement in 
OpenGL is going to be required?

Thanks and 73,

Bob W9RAN

cjcliffe, 2016-02-21:
@w9ran it's going to require some much needed OpenGL reworking which I'm going to try to do shortly after 0.2.0 beta release.

MSathishkumar1990, 2020-06-19:
(base) dinesh@dinesh-MS-7B24:~/Pangolin/build$ cmake ..
Build type not set (defaults to release)
-DCMAKE_BUILD_TYPE=Debug for debug
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find OpenGL (missing: OPENGL_opengl_LIBRARY OPENGL_glx_LIBRARY)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.10/Modules/FindOpenGL.cmake:369 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  src/CMakeLists.txt:169 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/dinesh/Pangolin/build/CMakeFiles/CMakeOutput.log".