cjcliffe/CubicSDR #227: Compile error under Fedora 23

digver, 2015-12-14:
Hello,
I'm trying to build with Fedora 23 and end up with this error.  Attached is a file of the entire output.
[make-errors.txt](https://github.com/cjcliffe/CubicSDR/files/61713/make-errors.txt)

In file included from /usr/include/wx-2.8/wx/memory.h:16:0,
                 from /usr/include/wx-2.8/wx/object.h:20,
                 from /usr/include/wx-2.8/wx/wx.h:16,
                 from /home/digger/src/git/CubicSDR/src/CubicSDR.cpp:7:
/usr/include/wx-2.8/wx/string.h:682:3: note: candidate: wxString::wxString(int) <near match>
   wxString(int);
   ^
/usr/include/wx-2.8/wx/string.h:682:3: note:   conversion of argument 1 would be ill-formed:
/home/digger/src/git/CubicSDR/src/CubicSDR.cpp:609:55: error: invalid conversion from ‘const char*’ to ‘int’ [-fpermissive]
     const wxString bwTitle("Set Demodulator Bandwidth");
                                                       ^
CMakeFiles/CubicSDR.dir/build.make:62: recipe for target 'CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o' failed
make[2]: *** [CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/CubicSDR.dir/all' failed
make[1]: *** [CMakeFiles/CubicSDR.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

This looks similar to #191 except that was chalked up to running out of memory. Is there any other info I can provide?

Thanks for any help,
digger

cjcliffe, 2015-12-14:
@digver It looks like you've got wxWidgets 2.8 -- you'll need 3.x minimum

digver, 2015-12-15:
@cjcliffe I wondered if it was something to do with wx versioning.

The naming scheme for wx in Fedora is wxGTK-devel and wxGTK3-devel.  I have both installed and I guess findWxWidgets doesn't go deep enough to pick up the right one. 

Running cmake doesn't throw any errors with both v2 and v3  installed. However if I erase wxGTK-devel, leaving wxGTK3-devel installed, cmake then ends with this --

64 bit compiler detected
CMake Error at cmake/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find wxWidgets (missing: wxWidgets_FOUND)
Call Stack (most recent call first):
  cmake/Modules/FindPackageHandleStandardArgs.cmake:374 (_FPHSA_FAILURE_MESSAGE)
  cmake/Modules/FindwxWidgets.cmake:868 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:111 (find_package)

-- Configuring incomplete, errors occurred!

If you have any ideas, I'll give them a try.

digver, 2015-12-16:
I found the answer. This bug marked as resolved addresses the issue of selecting and using the correct version of wxWidgets

https://cmake.org/Bug/view.php?id=15540

The short answer:
* In Fedora use the correct wx-config by running
    `cmake -DwxWidgets_CONFIG_EXECUTABLE=/usr/bin/wx-config-3.0 ../`

* For completeness maybe use FindwxWidgets.cmake from a newer version of cmake.

Here/s what is going on:

Fedora installs wx-config as wx-config-3.0. The FindwxWidgets.cmake supplied with CubicSDR only looks for wx-config. Newer versions of FindwxWidgets.cmake addresses this issue. But even then it will choose the first version in the list which is wx-config and never get to wx-config-3.0.

```cmake
find_program(wxWidgets_CONFIG_EXECUTABLE
      NAMES wx-config wx-config-3.1 wx-config-3.0 wx-config-2.9 wx-config-2.8
      DOC "Location of wxWidgets library configuration provider binary (wx-config)."
      ONLY_CMAKE_FIND_ROOT_PATH
      )
```
This shortcoming is overcome in Fedora by running cmake with 
`cmake -DwxWidgets_CONFIG_EXECUTABLE=/usr/bin/wx-config-3.0 ../`

However, in the end you can just run cmake as above -- without the newer FindwxWidgets.config -- and everything works.

My time availability comes and goes, but I can try and write up some Fedora specific install notes.

cjcliffe, 2015-12-16:
@digver thanks; with this info I think I have an idea of how I can update it to find/notify the wx version properly.

ab1jx, 2015-12-16:
Same problem with wxwidgets under Debian 8, which can use wx 2.8 or 3, I installed the dev deb for 3.0 and it gets past that.  Locate and whereis didn't find a wx-config so maybe there wasn't one installed. 

liquid/liquid.h it doesn't have a clue and I don't see anything in Synaptic.  I'm running 32-bit Debian BTW so your deb does me no good.  Googling takes me to https://github.com/lqd-io/liquid-sdk-ios/blob/master/Liquid/Liquid.h, is that the same Liquid?

OpenBSD (5.7) says: 
Scanning dependencies of target CubicSDR
[  1%] Building CXX object CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o
cc1plus: error: unrecognized command line option "-std=c++0x"
CMakeFiles/CubicSDR.dir/build.make:54: recipe for target
'CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o' failed
But it gets through cmake.  Using GCC 4.2.1, I also have 4.8.4

Windows (XP Pro) says it's not a valid Win32 application.

Dantali0n, 2015-12-17:
@ab1jx you can set to support windows xp by going in to ```project properties -> configuration properties -> platform toolset```. In this dropdown you will see a list of visual studio versions as of 2013 windows xp support is no longer included in compiled binarie you have to select ```visual studio 2013 - windows xp (v120_xp)```

cjcliffe, 2015-12-18:
@Dantali0n Thanks for the tip -- I'll try that out for the next 32-bit bundle.

closed 2016-04-02 by cjcliffe