cjcliffe/CubicSDR #228: Can we use PulseAudio sinks with RTAudio?

cjcliffe, 2015-12-15:
As suggested by e-mail; it's possible to create any number of audio sinks for input / output using pactl 'module-null-sink'.

Investigate and see if it's possible to list those sinks as audio outputs via some platform specific enumeration in RTAudio.

Relevant commands:
```
$ pactl load-module module-null-sink sink_name=CubicJT65-30M
$ pactl load-module module-null-sink sink_name=CubicJT65-20M
$ pactl load-module module-null-sink sink_name=CubicJT65-17M

So we get:-
pactl list | grep Cubic
	Argument: sink_name=CubicJT65-30M
	Argument: sink_name=CubicJT65-20M
	Argument: sink_name=CubicJT65-17M
	Name: CubicJT65-30M
	Monitor Source: CubicJT65-30M.monitor
	Name: CubicJT65-20M
	Monitor Source: CubicJT65-20M.monitor
	Name: CubicJT65-17M
	Monitor Source: CubicJT65-17M.monitor
	Name: CubicJT65-30M.monitor
	Monitor of Sink: CubicJT65-30M
	Name: CubicJT65-20M.monitor
	Monitor of Sink: CubicJT65-20M
	Name: CubicJT65-17M.monitor
	Monitor of Sink: CubicJT65-17M
```

edyirdaw, 2018-11-25:
Hello Charles,

1. I believe this would mean that cubicSDR would enable one to see perhaps a drop down list of previously created sink inputs and select which one to use instead of the default system sink input?

2. Do you have an approximate time estimate when this feature would appear in cubicSDR?

Thanks.

cjcliffe, 2019-01-04:
@edyirdaw yes I think it would provide a drop-down list; but I don't have any specific estimate for it at the moment as I haven't dug into RTAudio to see if it's possible yet.