cjcliffe/CubicSDR #230: cubicsdr : Depends: libwxgtk3.0-0 but it is not installable

Avamander, 2015-12-18:
I am running Ubuntu 15.10 and when I am trying to install the latest .deb package. dpkg says " cubicsdr : Depends: libwxgtk3.0-0 but it is not installable" my system has libwxgtk3.0-0v5 and libwxgtk3.0-0v5:i368. 

Would it be possible to use those packages instead?

cjcliffe, 2016-01-09:
Haven't had a chance to try but you should be able to force-install.   If you'd like to run an up-to-date version the Linux instructions are now somewhat complete for Ubuntu 15.10 which was previously problematic for some things.   If wxWidgets causes an issue there are also alternate instructions for building a local copy now too.

see: https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux

cjcliffe, 2016-01-14:
going to close this for now; future .deb releases will be fixed in a set of linux updates -- source build is recommended for the moment.

closed 2016-01-14 by cjcliffe