cjcliffe/CubicSDR #231: Ubuntu 15.10 display refresh issue.

cjcliffe, 2015-12-19:
Bug uncovered in Ubuntu 15.10 which is causing portions of the display to pause while hovered; guessing it's related to some focus/refresh issues.

cjcliffe, 2015-12-23:
Unable to reproduce with Nvidia driver in Ubuntu 15.10; I think I'll need to try an intel based system and possibly the window manager from the original report.

vsboost, 2015-12-23:
Hi Charles, I have Ubuntu 15.10 installed on my surface pro, is there anything you would like me to test?

cjcliffe, 2015-12-23:
@vsboost perhaps but I'm not quite sure what I'm testing for here yet; does it have integrated intel graphics and if so what chipset?

vsboost, 2015-12-23:
Intel HD Graphics 4400.

vsboost, 2015-12-23:
Every so often the display freezes for a split second its random, CPU usage is 94-107% this is using an sdrplay with 2,048 sample rate, switching the sample rate to 1,536 and the waterfall looks slightly jerky and CPU is around 50%, at 8mhz sample rate CPU is around 104%

cjcliffe, 2016-02-05:
Can't reproduce this specific error and original reporter has stated the GPU driver has additional problems which may make this unrelated.

closed 2016-02-05 by cjcliffe