cjcliffe/CubicSDR #234: SDRPlay API 1.8 available

Toontje, 2015-12-23:
See http://www.sdrplay.com/blog.html

Is it pretty much risk-free to install the latest API and build CubicSDR and all the SoapyStuff from scratch again to benefit from the latest enhancements in the API?

cjcliffe, 2015-12-23:
I think it's just the Windows API updated so far correct?   I'll have to give it a go and see if there's any problems but I think they have an update for SoapySDRPlay in the works as well.

frspin, 2016-01-14:
Now new 1.8.1 API is available.
Can we download it and recompile CubicSDR?

Regards
Franco Spinelli
IW2DHW

cjcliffe, 2016-01-14:
@frspin that's good news; I'd overlooked that it's on all platforms now -- I'll definitely make this a priority now that it's available.

cjcliffe, 2016-02-05:
latest APIs are building and running no problem, closing this up for now

closed 2016-02-05 by cjcliffe