cjcliffe/CubicSDR #235: Frequency offset not saved in Session file

Toontje, 2015-12-24:
Not there...
```
<?xml version="1.0" ?>
<cubicsdr_session>
    <header>
        <version>0.1.18-alpha</version>
        <center_freq>14238000</center_freq>
    </header>
    <demodulators>
        <demodulator>
            <bandwidth>6000</bandwidth>
            <frequency>73097000</frequency>
            <type>USB</type>
            <squelch_level>0.0207254</squelch_level>
            <squelch_enabled>1</squelch_enabled>
            <output_device>Apple Inc.: Built-in Output</output_device>
            <gain>0.310606</gain>
            <muted>0</muted>
        </demodulator>
        <demodulator>
            <bandwidth>5400</bandwidth>
            <frequency>14238001</frequency>
            <type>USB</type>
            <squelch_level>0.0207254</squelch_level>
            <squelch_enabled>1</squelch_enabled>
            <output_device>Jabra: Jabra UC VOICE 750a MS</output_device>
            <gain>1</gain>
            <muted>0</muted>
        </demodulator>
    </demodulators>
</cubicsdr_session>
```

cjcliffe, 2015-12-26:
This was actually removed on purpose because it conflicted with the new device settings saves (offset is saved per-device in regular or named configurations).

There will need to be some cleanup and rework before the named config could be swapped at runtime or saved to the session; for now you can use named configs to keep things separate, for OSX it's:
```
ccliffe$ /Applications/CubicSDR.app/Contents/MacOS/CubicSDR --config=alternate1
Creating new configuration file '/Users/ccliffe/Library/Application Support/CubicSDR/config-alternate1.xml' by copying '/Users/ccliffe/Library/Application Support/CubicSDR/config.xml'..ok.
Loading:: configuration file '/Users/ccliffe/Library/Application Support/CubicSDR/config-alternate1.xml'
```

Cheers!

Toontje, 2015-12-28:
Ok. Will use it like that.

Thanks!

Toontje, 2015-12-28:
This is what you get when you save a session.
```
<?xml version="1.0" ?>
<cubicsdr_session>
    <header>
        <version>0.1.18-alpha</version>
        <center_freq>7080000</center_freq>
    </header>
    <demodulators>
        <demodulator>
            <bandwidth>5400</bandwidth>
            <frequency>7133419</frequency>
            <type>LSB</type>
            <squelch_level>-100</squelch_level>
            <squelch_enabled>1</squelch_enabled>
            <output_device>Apple Inc.: Built-in Output</output_device>
            <gain>1</gain>
            <muted>0</muted>
        </demodulator>
    </demodulators>
</cubicsdr_session>
```
This is the generated session file.
```
<?xml version="1.0" ?>
<cubicsdr_config>
    <window>
        <x>-113</x>
        <y>-1057</y>
        <w>1920</w>
        <h>621</h>
        <max>0</max>
        <theme>0</theme>
        <snap>1</snap>
        <center_freq>7080000</center_freq>
        <waterfall_lps>5</waterfall_lps>
        <spectrum_avg>0.361702</spectrum_avg>
    </window>
    <devices>
        <device>
            <id>Generic RTL2832U :: 77771111153705700</id>
            <ppm>0</ppm>
            <offset>0</offset>
            <streamOpts>
                <buffers>15</buffers>
                <bufflen>16384</bufflen>
            </streamOpts>
            <settings>
                <direct_samp>0</direct_samp>
                <iq_swap>false</iq_swap>
                <offset_tune>false</offset_tune>
            </settings>
        </device>
        <device>
            <id>Generic RTL2832U OEM :: 00000001</id>
            <ppm>0</ppm>
            <offset>0</offset>
            <streamOpts>
                <buffers>15</buffers>
                <bufflen>16384</bufflen>
            </streamOpts>
            <settings>
                <direct_samp>0</direct_samp>
                <iq_swap>true</iq_swap>
                <offset_tune>false</offset_tune>
            </settings>
        </device>
        <device>
            <id>SDRPlay RSP</id>
            <ppm>0</ppm>
            <offset>-120</offset>
            <streamOpts />
            <settings>
                <if_mode>Zero-IF</if_mode>
                <use_low_if>true</use_low_if>
            </settings>
        </device>
    </devices>
</cubicsdr_config>
```
Why not save all the data into one session/config file? Like this you can create use case based session files.

cjcliffe, 2015-12-28:
For now the config and session are separate entities; they can't both contain device settings at the moment because device settings are not interchangeable at runtime and won't be until some related issues such as refreshing devices is also resolved -- in the future it should be possible to save device settings but we'd need to decide if that's even a thing we should do and what settings make sense.

For this specific issue we actually need to debate whether the offset needs to be saved per-device per-session.  

My primary argument is that to keep the offset with the per-device configuration is simply because with my setup here I move up-converters between devices; after changing devices I would be required to load, reset the offset and re-save each related session file with the updated device offset.  Moving additional device settings into the session would further fragment the ability to maintain the current configuration state in one place.

Additionally if I save the offset with the session file and disregard the device; jumping between multiple devices with different up-converter configurations and loading sessions would result in that offset needing to be adjusted each time the session was loaded for a different device.

As it stands now I can set the up-converter offset once (per-configuration file) and any sessions adhere to the current device configuration and there's no chance of fragmentation or mismatching device settings.

If you're simply attempting to use offset as 'PPM' adjustment that's partly my bad -- it needs to be implemented in SoapySDRPlay as it's currently missing -- I just haven't had the need to adjust PPM on my SDRPlay so far it's been spot on :)

cjcliffe, 2016-01-08:
Going to close this one up, my rant was based on some false assumptions of the use case here (saving IF offsets I believe); I think we're going to cover this off in #70 and it's business as usual.

closed 2016-01-08 by cjcliffe