cjcliffe/CubicSDR #236: Question: How to use SoapyRedPitaya with CubicSDR?

dc1rdb, 2015-12-25:
Setup:
RedPitaya on LAN at address 192.168.178.32
latest SoapyRedPitaya module active:
```
imac:~ rduering$ SoapySDRUtil --probe="driver=redpitaya, addr=192.168.178.32"
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device driver=redpitaya, addr=192.168.178.32

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=redpitaya
  hardware=redpitaya

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 1 Tx
  Timestamps: NO

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CF32
  Native format: CF32 [full-scale=1]
  Full gain range: [0, 0] dB
  Full freq range: [0.05, 60] MHz
    RF freq range: [0.05, 60] MHz
  Sample rates: [0.02, 1.25] MHz

----------------------------------------------------
-- TX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CF32
  Native format: CF32 [full-scale=1]
  Full gain range: [0, 0] dB
  Full freq range: [0.05, 60] MHz
    RF freq range: [0.05, 60] MHz
  Sample rates: [0.02, 1.25] MHz
```
CubicSDR device selection does not detect the device. Typing in the LAN address via "Add remote" results in the following console output:

```
SoapySDR init..
	API Version: v0.4.0-gb07f54d3
	ABI Version: v0.4-2
	Install root: /usr/local
	Loading modules... 
Checking local system SoapySDR modules..Loading bundled SoapySDR module /Applications/CubicSDR.app/Contents/MacOS/modules//libHackRFSupport.so..
Loading bundled SoapySDR module /Applications/CubicSDR.app/Contents/MacOS/modules//libairspySupport.so..
Loading bundled SoapySDR module /Applications/CubicSDR.app/Contents/MacOS/modules//libbladeRFSupport.so..
Loading bundled SoapySDR module /Applications/CubicSDR.app/Contents/MacOS/modules//libremoteSupport.so..
Loading bundled SoapySDR module /Applications/CubicSDR.app/Contents/MacOS/modules//librfspaceSupport.so..
Loading bundled SoapySDR module /Applications/CubicSDR.app/Contents/MacOS/modules//librtlsdrSupport.so..
Loading bundled SoapySDR module /Applications/CubicSDR.app/Contents/MacOS/modules//libsdrPlaySupport.so..
Loading bundled SoapySDR module /Applications/CubicSDR.app/Contents/MacOS/modules//libuhdSupport.so..
Mac OS; Clang version 6.1.0 (clang-602.0.53); Boost_105900; UHD_003.009.000-MacPorts-Release

	Available factories...airspy, bladerf, hackrf, null, redpitaya, remote, rfspace, rtlsdr, sdrplay, uhd
mir_sdr_Init: starting hardware initialization
mir_sdr_Init: gR=40dB fs=2.048MHz rf=222.064MHz bw=1.536MHz if=0.000MHz

mir_sdr_usb_USB DLL: Revision 1.8.0

mir_sdr_usb_Init: Timeout expired/failed to establish connection with the device

mir_sdr_2500_Init: mir_sdr_usb_Init() failed
mir_sdr_Init: mir_sdr_2500_Init() Error 1

Reporting enumeration complete.
SDR enumerator done.
SDR enumerator starting.
mir_sdr_Init: starting hardware initialization
mir_sdr_Init: gR=40dB fs=2.048MHz rf=222.064MHz bw=1.536MHz if=0.000MHz

mir_sdr_usb_USB DLL: Revision 1.8.0

mir_sdr_usb_Init: Timeout expired/failed to establish connection with the device

mir_sdr_2500_Init: mir_sdr_usb_Init() failed
mir_sdr_Init: mir_sdr_2500_Init() Error 1

Enumerating remote address: 192.168.178.32:1001
[ERROR] SoapyRemote::find() -- transact FAIL: SoapyRPCUnpacker::recv(header) FAIL: recv() [54: Connection reset by peer]

Reporting enumeration complete.
SDR enumerator done.
```

guruofquality, 2015-12-25:
FYI: I put a bug on the SoapyRedPitaya issue tracker, I think this is the problem: https://github.com/pothosware/SoapyRedPitaya/issues/3

cjcliffe, 2015-12-26:
Hmm, I should probably add some sort of option to the "add device" dialog as well for setting up devices that might not fit the default auto-discovery.

guruofquality, 2015-12-27:
> Hmm, I should probably add some sort of option to the "add device" dialog as well for setting up devices that might not fit the default auto-discovery.

Thats probably true. There will be cases where discovery is not possible or not functional. In this particular case, I think that the user-specified args should still make it through enumeration, so if the user specified an address, findSoapyRedPitaya() would test connect and return the actual set of args used for device instantiation.

cjcliffe, 2016-01-20:
@dc1rdb 'soapysdr-manual-device' branch is now available if you want to try manual device specifications; doesn't save yet but will soon.  

The 'Add Remote' is replaced with an 'Add' button (which defaults to remote option) but you should be able to select 'redpitaya' module and enter 'addr=192.168.178.32' as the parameters.

dc1rdb, 2016-01-20:
Hi Charles, many thanks! Currently, I don't have access to my RedPitaya. Will test as soon as I get it back.

cjcliffe, 2016-01-22:
@dc1rdb Manual device specification is now in; closing issue for now but let me know how it works out with RedPitaya when you get a chance :+1:

dc1rdb, 2016-01-22:
Will do as soon as I get my board back! Thanks - much appreciated!

dc1rdb, 2016-01-24:
Hi Charles - great job, works well with CubicSDR v0.1.22 and current SoapyRedPitaya. All it takes is entering the local address of RedPitaya at the SDR selection window. Audio is a bit choppy at 1.25 MHz sampling rate, however. Not sure if this is a SoapyRedPitaya or network or CubicSDR issue. Any idea how I can further test this?

dc1rdb, 2016-01-24:
A very minor observation: with SoapyRedPitaya selected, CubicSDR settings pull-down menu should not have an entry for "Automatic Gain", because RedPitaya has no gain control at all.

cjcliffe, 2016-01-24:
@dc1rdb hmm will have to hide that option in that case then; hadn't considered that a device might not have gain control.   Not sure how to test much further here without hardware; could be Network, SoapyRedPitaya or CubicSDR issue -- I'm guessing you can eliminate CubicSDR and Network by trying some other applications and network connections -- if it doesn't improve we should probably look at SoapyRedPitaya.

cjcliffe, 2016-02-29:
With latest commit gain options should now be hidden for Red Pitaya or any device without gain option.

closed 2016-02-29 by cjcliffe