cjcliffe/CubicSDR #237: Need help with a compiling error!

Spruce8181, 2015-12-26:
I've successfully downloaded Microsoft Visual Studio. I've extracted the Widgets file into a folder.

After the step below I get 24 errors on the Build Solution step. 

Their all related to not being able to find a CL.exe file.

---------------------------------------------------------------------------------------------------------------------
Navigate to C:\MSVCDev\wxWidgets-3.0.2\build\msw\ (or wherever you extracted) and open wx_vc12.sln, Click "OK" to update the project on load.

Choose "Release" and "x64" for the build configuration and "Build Solution". All should compile successfully and you can close the project.
---------------------------------------------------------------------------------------------------------------------

Please advise on how to fix this issue. What have I done wrong?

cjcliffe, 2015-12-26:
Sorry about that -- I probably should have made it clear on the build instructions that you need to install the C++ compiler; sounds like it just wasn't configured during the visual studio install.

Run the installer again and make sure you've selected "C++" under the programming languages section; you should be able to proceed after that.   I'll update the instructions as well.

Spruce8181, 2015-12-27:
thank you very much

cjcliffe, 2015-12-29:
Not a problem, glad that worked.

closed 2015-12-29 by cjcliffe