cjcliffe/CubicSDR #239: No sideband rejection for SSB

cminyard, 2015-12-28:
I got an SDRPlay for Christmas and I've been playing with the various radios.  CubicSDR is the only one I could get to even marginally work, and for the most part it works.  However, it appears I get no sideband rejection at all with SSB.  If I set USB and put a signal in the passband, and then I switch to LSB, I get no discernible or measurable different.  And vice versa, and moving the passband around so the signal moves into the other sideband.

I played around with the code a bit.  I know a little about DSP, some about radio, and a lot about writing software.  I added a Hlibert transform for the USB and LSB modes in liquid-dsp, and that made about 30-40db drop from the sideband.  Looking in CubicSDR, it appears it has a resamp2 filter and it appears to have correct parameters, but it does not appear to be working.  30-40db is not terribly satisfactory, either.

However, why is the filter applied before the demodulation?  It appears the filter should cover 0..bw/2 (or -bw/2..0), but it is applied before the demodulation, so I don't think it would really do much.  But I don't understand this completely.

Does anyone else see this, or am I just doing something wrong, or crazy?

Why doesn't liquid-dsp implement the filter in it's module?  Yeah, I know, ask on the liquid-dsp site.

cjcliffe, 2015-12-28:
@cminyard there should already be rejection in the particular resamp2 function used; my original version used a hilbert transform, FIR and shift which was a bit better but required many more steps and CPU.  You can also simply increase the rejection with the last parameter to resamp2 which is currently 60db.

To understand what it's doing you need to know the filter is doing a bit more than standard resamp2 resampling -- using the special function 'resamp2_crcf_filter_execute' it should already be performing a hilbert transform and frequency shift to split the upper and lower bands into two separate outputs (x & y) and center them across the baseband -- LSB uses x and USB uses y.

Some additional filtering performed on x or y outputs [-0.25..0.25] before demodulation would definitely improve rejection; I've added it to #214 as the first noise reduction issue to solve.

cminyard, 2015-12-29:
I understand all this, but from what I can tell, it's not working.

All of this is on LSB.

If I set the audio display to show power at frequency and park it on an empty area of spectrum, it shows around -200dB, which seems very low for background noise, but that's what it says.

If I set it so a narrowband source of noise is at around 1000Hz in the passband, the audio power shows around -100dB.  If I switch to USB, I get exactly the same level.  If I set it back to LSB and move the frequency to the right of the passband, I get -100dB.

If I enable my hilbert transform I get different results.  If I set it to LSB and put it at 1000Hz in the passband, I get -100dB.  If I switch to USB i get -130dB.  Switch back to LSB and put the signal to the right of the passband, -130dB.  And in the last two, I can hear the background noise come way up and the signal is still present in the audio, just more faint, as the gain is automatically increased.

So unless I'm not understanding something, it appears to not be working at all.

A couple of more things on your statements.  My understanding of SSB rejection using phase is that you have to do the hilbert transform after the downconversion because you only do it on I or Q, not on both.

Also, if you are running the fresamp2_crcf filter before you downconvert, wouldn't it have to be centered around the frequency the signal is at before you downconvert?  My understanding is that it is centered around zero.

I'm still learning, so there's a lot I don't understand.  Thanks for your previous post

cjcliffe, 2015-12-29:
@cminyard it's possible it's not filtering as I expected; I was mostly instructed on that method as an alternate from the author but it may have been intended that I still perform some level of side-band filtering; I'll have to re-investigate.

Sorry my resamp2_crcf_filter_execute description was a bit unfinished; basically if you mix the LSB or USB signal so that it's centered across baseband (-/+0.25) and then perform decimation then re-interpolate and mix back to baseband you'll get the upper or lower bands (x & y) that function produces; in my previous version I did something similar to that but included an additional hilbert and FIR filter (which was not specific to sideband).

As far as actual resampling to the audio bandwidth, that's performed after LSB / USB demodulation by the ```buildAudioOutput(akit, audioOut, true);``` command.  You can disable the automatic audio gain by setting the last parameter to false.

cjcliffe, 2016-01-02:
@cminyard I've done some more testing here on my HF antenna and the sideband rejection is definitely lacking; most likely something changed during the latest modem conversions -- if you have a working solution ready you'd like to submit I'd pull it in otherwise I'll take a look at reworking the filtering soon.

Thanks!

cminyard, 2016-01-02:
On 01/01/2016 09:11 PM, Charles J. Cliffe wrote:
> @cminyard <https://github.com/cminyard> I've done some more testing
> here on my HF antenna and the sideband rejection is definitely
> lacking; most likely something changed during the latest modem [...]

My current solution isn't that great, a filter should do a better job 
than the phasing solution I did.  I'm not quite sure how to do a filter, 
though I could try.

Thanks!

frspin, 2016-01-04:
I can confirm this bug in normal use.
If I am on USB and a signal is present in LSB portion I can heard it.

Regards
Franco Spinelli
IW2DHW

cjcliffe, 2016-01-05:
@frspin thanks for the confirmation; I have a few simple ideas on how to improve the filtering which I'll try soon.

cminyard, 2016-01-05:
On 01/04/2016 08:04 PM, Charles J. Cliffe wrote:
>
> @frspin <https://github.com/frspin> thanks for the confirmation; I 
> have a few simple ideas on how to improve the filtering which I'll try 
> soon.
>

IMHO the right way to fix this is in liquid-dsp.  It would be a lot 
easier to handle this in the IQ domain at the demodulation.  I'm not 
really sure you can fix this once you demodulate to the real domain.

If you filter the data before you demodulate it, that would work, too, I 
think.  That's what is sort of there already, but I think it's at the 
wrong frequency.  But still, it's kind of silly for liquid-dsp to offer 
the USB and LSB options and do absolutely no filtering.  If that's the 
case it should only offer DSB.

-corey

cjcliffe, 2016-01-05:
@cminyard yeah I think most of the fix is going to be correcting my usage which is first to try; plus adding some additional resamp2 pre-filtering can't hurt -- I can even make it a configurable option with the new modem properties.

cjcliffe, 2016-01-10:
@cminyard I've fixed side-band suppression by removing resamp2 and just using a simple shift+fir filter similar to my original solution awhile back.  

I'll close this one up for now but feel free to make additional tweaks / improvements / suggestions.

closed 2016-01-10 by cjcliffe