cjcliffe/CubicSDR #242: Raspberry Pi2

Peter-H-G, 2016-01-01:
Trying to install on Pi2 _ Debian
At liquid_dsp after sudo make install the libliquid.so and libliquid.a are not found.

cjcliffe, 2016-01-02:
@Peter-H-G if 'sudo make install' succeeded for liquid-dsp then you may just need to run 'sudo ldconfig' or reboot to recognize the new library.  Let me know if that helps and I'll update the instructions accordingly.  Not completely sure if it will actually run on Pi2 yet; others have attempted but ran out of swap -- so I'd suggest making sure you have a swap file enabled :)

cjcliffe, 2016-01-07:
Pi2 has arrived here today; should be able to make some progress on wiki instructions soon.

Peter-H-G, 2016-01-08:
Dear Charles,
this is good news.
What I try to install is: 
https//github.com/cjcliffe/CubicSDR/wiki/Build-Linux
following the paper I get problems in "Build and Install liquid-dsp" not 
finding libliquid.so and libliquid.o
in attachment is the copy of my inputs.

On my old ASUS u5a under XP-SP3 it works with the SDRPlay hardware.

To explain what I try to do: Measuring FM and TV- Transmitters having 
problems with antennas. It happens often that feeders between splitters 
and the antenna- panels have problems. As the RF-field recomposes in a 
distance of 10 to 20 wavelengths, this measurement must be done very 
close to the antennas.
Te new legislation authorises drones only up to 2,5 kg and an additional 
charge of maxi 1 kg.
a shielded Raspberry with a SDRPlay together could do the business. On 
thing is still missing: I need a additional info of the field-strength. 
This signal will be injected to the downlink-channel of the 
telecommand-system (JETI) which has already GPS and barometric height. 
The barometric height has a precision of 50 cm. But there is no need to 
have absolute values, the differential values are good enough.
Please keep it secret for the moment.
Attachments date: day month year and excuse my faults in English.
Thank you for your ideas
Peter

cjcliffe, 2016-01-08:
Hi Peter, just note when you reply to the GitHub e-mails they post to public GitHub automatically; I've edited your previous comment online.

Please re-send the attachments and question to cj@cubicproductions.com I can try to help you with the details there.   I'll post additional information about compiling liquid-dsp (which I managed to do last night) on this GitHub issue shortly.

cjcliffe, 2016-01-21:
At the moment it looks like this is going to require an OpenGL ES update and possibly a custom GL canvas initialization.   It almost-sorta-works with 'glshim' project but the windowed displays each just flash briefly full-screen and some required commands aren't implemented.

cjcliffe, 2016-01-21:
Going to make OpenGL ES / 3.x a priority after 0.2.0 release; I think the next major performance boost will be the result of moving the CPU graphics work to GPU shader programs as well (which will definitely help the Pi).

cjcliffe, 2016-03-21:
It appears that wxWidgets 3.1.0 now includes OpenGL ES (and OpenGL 3.x+ for desktop) support for wxGLCanvas -- I'm going to experiment with enabling OpenGL ES support for Pi and see what happens :)

vsboost, 2016-03-21:
Good luck, ill be sure to test it on my Rpi3

Peter-H-G, 2016-03-22:
Thank you for that info.
In the meantime there is a very simple thing, based on a Phillips - 
TunerFM 1216 making FM and TV and is is piloted by an Arduino. Outputs 
are Fieldstrength, and audio.
But there is still the idea to make it with the RPi2.

Toontje, 2016-03-22:
I was brave enough to compile wxWidgets and CubicSDR onmy Raspberry Pi B+
only to find out that the B+ is not fast enough to run CubicSDR in any
reasonable way.
Compiling CubicSDR and wxWidgets took over 4 hours each. ;-)

Peter-H-G, 2016-03-22:
OK the RPi2 is twice faster, but as you tried it out, this is not the 
good solution. Thank to heaven I had both ways in my mind.
I use the "counter-reaction" which regulates the IF-frequency - 
amplification in the Tuner FM1216 as an indicator for field-strength. An 
it works at least now for FM; my little thing has just 254gram and 
"eats" 220mA on 5V. Since even professional drones are now limited to 
2,5 kg plus 400 g payload in France.
And I think they are right if you think to the last encounter of a 
LH-Airbus A380 near LA in a height of 1500m.
Thank you for cooperation.

cjcliffe, 2016-03-22:
Trying to get things going on the RPi2 right now; so far no luck just flipping on the ES2 flag but OpenGL support looks to be usable up to OpenGL ES 2.x / OpenGL 4.5 in wxWidgets 3.1.0 so I'm hopeful still.

On the RPi2 you can at least add -j3 to the 'make' commands to use a couple additional cores to reduce compile times -- there's some nasty GCC warnings from the CubicVR2 includes that are clogging the terminal that I'm trying to resolve as well which should help.

cjcliffe, 2016-06-09:
Haven't tried this on Pi2 yet but it may be working on the latest Raspbian update (On Pi3 at least) according to this:

https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=150754

From the post:
```
==============================================
# Building CubicSDR for Linux
# This script will successfully build CubicSDR under Debian Jessie on the Raspberry Pi3.
# Adapted from https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux
# Larry Dighera June 7, 2016
# LDighera@att.net

apt-get install git build-essential automake cmake
apt-get install libpulse-dev libgtk-3-dev
apt-get install freeglut3 freeglut3-dev
mkdir cubicsdr && cd cubicsdr
git clone https://github.com/pothosware/SoapySDR.git
cd SoapySDR
mkdir build && cd build
cmake ../ -DCMAKE_BUILD_TYPE=Release
make -j4
make install
ldconfig
SoapySDRUtil --info #test SoapySDR install
cd ..
git clone https://github.com/jgaeddert/liquid-dsp
cd liquid-dsp
./bootstrap.sh
./configure --enable-fftoverride 
make -j4
make install
ldconfig
cd ..
wget https://github.com/wxWidgets/wxWidgets/releases/download/v3.1.0/wxWidgets-3.1.0.tar.bz2
tar -xvjf wxWidgets-3.1.0.tar.bz2  
cd wxWidgets-3.1.0/
mkdir -p ~/Develop/wxWidgets-staticlib
./autogen.sh 
./configure --with-opengl --disable-shared --enable-monolithic --with-libjpeg --with-libtiff --with-libpng --with-zlib --disable-sdltest --enable-unicode --enable-display --enable-propgrid --disable-webkit --disable-webview --disable-webviewwebkit --prefix=`echo ~/Develop/wxWidgets-staticlib` CXXFLAGS="-std=c++0x" --with-libiconv=/usr
make -j4 && make install
cd ..
apt-get install libhamlib2 libhamlib-utils libhamlib++-dev libhamlib-dev libhamlib-doc python-libhamlib2
git clone https://github.com/cjcliffe/CubicSDR.git
cd CubicSDR
mkdir build && cd build
../ -DUSE_HAMLIB=1 -DCMAKE_BUILD_TYPE=Release -DwxWidgets_CONFIG_EXECUTABLE=~/Develop/wxWidgets-staticlib/bin/wx-config
make
cd x86/
./CubicSDR
make install
cd ..
git clone https://github.com/pothosware/SoapyRTLSDR.git
cd SoapyRTLSDR
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release
make
make install
ldconfig
# should now show RTL-SDR device if connected
SoapySDRUtil --probe     

# Add CubicSDR to the X11 menu with the Menu Editor: cubicsdr/CubicSDR/build/x86/CubicSDR
```

LDighera, 2016-06-11:
I was surprised to find that CubicSDR would run on the Raspberry Pi3.  Although a bit slow, and tending to overheat the Pi, it looks to be usable.  It was necessary to reduce the Sample Rate to prevent buffer overflow.

The present issue is getting CubicSDR audio out.  It appears to possibly be the result of the current (4.4.11-v7+) Debian Jessie release default configuration of PulseAudio.  This is apparently currently under active development from the information here: https://www.raspberrypi.org/blog/another-update-raspbian/ .  Killing and restarting the PulseAudio daemon successfully produced CubicSDR Audio: /bin/sh /usr/bin/start-pulseaudio-x11 .

freebil, 2016-08-03:
Any news about this?

LDighera, 2016-08-03:
Perhaps the information at this page will help: [https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=150754&p=1017039#p1017039](url)

cjcliffe, 2016-08-04:
@freebil the solution is currently as @LDighera has suggested at https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=150754 -- until I do some updates to the OpenGL graphics code (which wxWidgets is GL canvas slowly catching up to) and some additional optimizations it's not going to be as performant as it could be.  

This issue will stay open and I'll eventually post some things to test on Pi here when relevant optimizations are implemented.

closed 2017-02-28 by Peter-H-G