cjcliffe/CubicSDR #243: Setting Frequency offset causes a continual frequency shift

ddb, 2016-01-03:
Using the code from commit 693e67542186cd48b00375b0ff7068d2c3842533

If you set a frequency offset that is non-zero, the program will continually apply that offset, causing the center frequency to continually drift in the direction of the offset.

I'm using an SDRPlay RSP1, running on OS X 10.11.2

If this cannot be reproduced, I can provide a movie of the effect.

ddb, 2016-01-03:
Setting the offset to 0 eliminates the drift.

ddb, 2016-01-03:
Problem seems to be in this commit: c94cdb085594f0be5390e7c4ab442da7cf2737c5

I'm not familiar with the code, so this next part could be completely misinformed and incorrect, and I apologize in advance.

I think the bug is contained in the following code:

```
@@ -242,6 +245,11 @@ void SDRThread::updateSettings() {
         freq_changed.store(false);
     }

+    double devFreq = device->getFrequency(SOAPY_SDR_RX,0);
+    if (devFreq != frequency.load()) {
+        wxGetApp().setFrequency((long long)devFreq);
+    }
+
     if (agc_mode_changed.load()) {
         SDRDeviceInfo *devInfo = deviceInfo.load();

```

I suspect that reading the frequency and then checking for a difference without taking the frequency offset into account is the issue here.

cjcliffe, 2016-01-03:
@ddb hah, yes that's a hilarious side-effect of forgetting about the offset when I made that adjustment :)

I'll fix that up right away.

cjcliffe, 2016-01-03:
Ok, fix for offset drifting is committed to master, cheers!

closed 2016-01-03 by cjcliffe

ddb, 2016-01-03:
Confirmed.

Awesome response.  Thanks!