cjcliffe/CubicSDR #245: Mouse hover displaying demodulator frequency

Toontje, 2016-01-06:
Is it possible that when you move your mouse over the waterfall or spectrum display that somewhere, preferable close to the mouse pointer, but any other location on the screen would be good as well, to show the frequency the mouse is hovering over?

cjcliffe, 2016-01-13:
Now committed to master while hovering waterfall + snap placement preview :)

jdeitch, 2016-01-13:
@cjcliffe Would it be possible to move the TUNED freq display up one line and maybe change the color?  When you hover close to the tuned freq the mouse pointer frequency the two frequencies collide on the display making it very hard to read both.

cjcliffe, 2016-01-13:
@jdeitch sure that makes sense -- I'll take a look and what I can do to make overlapping them more friendly.

cjcliffe, 2016-01-14:
@jdeitch fixed up now with your suggestions -- also integrated the indicator scheme used on the waterfall which makes it more apparent what's going on when you're about to add a demodulator (green = new, yellow = move, red = to be moved) which should be good for new users too.

closed 2016-01-14 by cjcliffe

jdeitch, 2016-01-14:
@cjcliffe Looks good.  Thanks!