cjcliffe/CubicSDR #246: Option to set Center frequency to current demodulator frequency

Toontje, 2016-01-06:
Use case: I am using the SDR to search for stations. For this i need at least one demodulator. I can move this demodulator around. When i find a station i want to work i need to set the transceivers frequency to the modulator frequency. A key or button would be nice to achieve this.

cjcliffe, 2016-01-06:
This seems pretty simple, I think in general just hovering a demod and pressing 'c' would be a logical way to center it.  Combined with #245 this would make a simple way to set the center frequency to the hovered frequency as well (when not hovering over a demodulator).

Toontje, 2016-01-07:
Would do it for me.

cjcliffe, 2016-01-13:
Now committed to master:
- Hovering the waterfall and pressing 'c' will now center on hovered frequency (obeys snap setting)
- Hovering a demodulator on waterfall (or upper left) and pressing 'c' will center frequency on it.

closed 2016-01-13 by cjcliffe