cjcliffe/CubicSDR #247: Rig Control/Model defaults to Hamlib Dummy

Toontje, 2016-01-06:
Doesn't retain the last model i selected.

cjcliffe, 2016-01-06:
@Toontje hmm, strange it should save it unless you crashed instead of quit :)  I always avoid saving the configuration until a successful quit in case the crash was settings-related.

I'll take a look and check if it's doing that right.

Toontje, 2016-01-07:
It keeps the correct configuration when the SDR device selection screen comes up when you first start CubicSDR, but this setting doesn't propagate through to the Rig Control/Model menu. It's only the model that doesn't get saved. The Serial Rate and Control Port is retained correctly.

cjcliffe, 2016-01-07:
The SoapyAudio rig control and hamlib implementation is now separate from CubicSDR so they're not connected -- you can technically have a rig controlling another rig now through SoapyAudio ;)

The one in the menu should save independently; that should be a quick fix -- it may just be not setting the checkbox selection properly.

cjcliffe, 2016-01-10:
Now fixed, was just a typo on config load

closed 2016-01-10 by cjcliffe

Toontje, 2016-01-16:
Confirmed! Awesome work this week!!