cjcliffe/CubicSDR #248: CubicSDR: The Manual

cjcliffe, 2016-01-07:
- [ ] Introduction
  - [ ] Quick overview of SDR and what CubicSDR does
  - [ ] Basic use cases
  - [ ] Example / tutorial / video links around the web
- [ ] "Getting Started" with installation / build links.
  - [ ] Current status and description/credit for CubicSDR's dependencies
  - [ ] SoapySDR introduction
- [ ] SDR Device selection
  - [ ] Device properties
  - [ ] Starting device, sample rate control
- [ ] Main Application Window - functionality and usage
  - [ ] Annotated image, main window layout description, general overview
  - [ ] Quick "cheat-sheet" table of available functions / keyboard commands with descriptions
    - [ ] Printable PDF download version
  - [ ] Main spectrum
  - [ ] Waterfall
  - [ ] Tuning bar
  - [ ] Demodulator spectrum and waterfall
  - [ ] Squelch
  - [ ] Scope / Visuals
  - [ ] Color Themes
  - [ ] Sessions
  - [ ] Audio controls and piping
  - [ ] Misc. features, tips, tricks
- [ ] Advanced / specific use case features
  - [ ] Using Manual Gain control
  - [ ] Using Rig control
  - [ ] Using SoapyAudio
  - [ ] Using SoapyRemote
  - [ ] Using JACK Audio for piping
  - [ ] Using PulseAudio routing / dummy input tools

Dantali0n, 2016-01-08:
Do you want to use the github wiki as manual or did you have other ideas such as a external wiki maybe on the cubicsdr site?

cjcliffe, 2016-01-08:
@Dantali0n hadn't decided yet; I was thinking of using the wiki as "development" mode to do it in markdown and possibly export the "stable" versions of each manual to the main website site in static format.

NN4F, 2016-01-09:
Any Help you need to the manual let me know if I can help....

cjcliffe, 2016-01-09:
@NN4F thanks!  I'll hopefully start to fill in some content in the Wiki for the first 'alpha' and it would be great if you could help translate my developer speak and terminology to the amateur radio community terms as well to make sure everyone will understand (and vice-versa for me ;-)  Expanding upon content and giving user experience examples once we get started would be welcome as well.

NN4F, 2016-01-09:
Sure, count me in Charles....even though I'm a new user and learning more day by day... :)

vsboost, 2016-01-09:
Ill contribute for sure.

cjcliffe, 2016-01-09:
@NN4F @vsboost thanks! I'll probably be working on the outline for a few iterations; feel free to comment or suggest any changes as I go.

henrybehmann, 2016-01-27:
Hi Charles,

Please ignore my previous message.I have found the offset setting. All I was seeing in the menu selection was offset tune. I didn't see the first item in the menu. Sorry about that. I am getting olda nd this is what happens. I am looking forward to the manual so I can make use of all the features you have in this program. Now I will try for HF and see what I can catch.

I very much appreciate all the effort and dedication that went into CubicSDR. I like it better than the other SDR software I have tried. I look forward to saving frequencies from a list so I don't have to enter that every time I monitor a different station.

Thanks and cheers, Henry

vsboost, 2016-02-13:
Hi Charles,

Any chance you could include somewhere in the wiki a list of all the short cut keys?

Cheers

Toontje, 2016-02-13:
There is a Hover Tips setting now which helps you finding the right keys in the right situation.

cjcliffe, 2016-02-13:
Aye, added to the list -- 0.2.0 milestone is shrinking so I'll need to get a first draft ready for release soon :)

theforcedk, 2016-04-03:
I like Read The docs for writing documentation (supports markdown) because it's very flexiable (and free).. you probably know it, but check out a Ruby doc page here: http://gallium.readthedocs.org/en/latest/  - more "inspiration" here: https://readthedocs.org/

neilwillgettoit, 2016-05-20:
@cjcliffe so do you mind if I start making github wiki pages ?

cjcliffe, 2016-05-20:
@neilwillgettoit that'd be ok with me -- I'm not 100% sure on the layout or format yet though; that's what #348 is going to help decide.

Dantali0n, 2016-05-20:
I really like readthedocs and the way it looks, especially the way codeigniter did it [https://www.codeigniter.com/user_guide/](https://www.codeigniter.com/user_guide/)

cjcliffe, 2016-05-20:
Yeah I think ReadTheDocs is looking like a good option; I'm going to start listing the choices over in #348