cjcliffe/CubicSDR #254: xcb] Unknown request in queue while dequeuing

TheOncomingStorm, 2016-01-13:
I'm using Gentoo 64-bit. 

Full error is:
```
./build/x64/CubicSDR
SDR post-processing thread started..Spectrum visual data thread started.

Spectrum visual data thread started.
FFT visual data thread started.

Audio Device #0 PulseAudio
        Default Output? Yes
        Default Input? Yes
        Input channels: 2
        Output channels: 2
        Duplex channels: 2
        Native formats:
                16-bit signed integer.
                32-bit signed integer.
                32-bit float normalized between plus/minus 1.0.
        Supported sample rates:
                8000hz
                16000hz
                22050hz
                32000hz
                44100hz
                48000hz
                96000hz

Available vertical sync SwapInterval functions: 
        glxSwapIntervalEXT: No
        DRI2SwapInterval: No
        glxSwapIntervalMESA: No
        glxSwapIntervalSGI: No
No vertical sync swap interval functions available.
Loaded font 'Bitstream Vera Sans Mono' from '/home/user/src/CubicSDR/build/x64/vera_sans_mono16.fnt', parsed 167 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/home/user/src/CubicSDR/build/x64/vera_sans_mono12.fnt', parsed 167 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/home/user/src/CubicSDR/build/x64/vera_sans_mono18.fnt', parsed 167 characters.
SDR enumerator starting.
SoapySDR init..
        API Version: v0.4.0-gd17177f3
        ABI Version: v0.4-2
        Install root: /usr
        Loading modules... 
        Available factories...nullJust 'null' factory found.
[xcb] Unknown request in queue while dequeuing
[xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
[xcb] Aborting, sorry about that.
CubicSDR: /var/tmp/portage/x11-libs/libX11-1.6.3/work/libX11-1.6.3/src/xcb_io.c:179: dequeue_pending_request: Assertion `!xcb_xlib_unknown_req_in_deq' failed.
Aborted

```

http://stackoverflow.com/questions/31952711/threading-pyqt-crashes-with-unknown-request-in-queue-while-dequeuing
This suggests it's an issue with XInitThreads needing to be specifically set to be threadsafe in PyQt.

TheOncomingStorm, 2016-01-14:
Here's a backtrace if that helps. 

```
gdb ./build/x64/CubicSDR
GNU gdb (Gentoo 7.10 vanilla) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://bugs.gentoo.org/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./build/x64/CubicSDR...(no debugging symbols found)...done.
(gdb) run
Starting program: /home/eric/src/CubicSDR/build/x64/CubicSDR 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[New Thread 0x7fffe9f43700 (LWP 19440)]
SDR post-processing thread started..
[New Thread 0x7fffe9742700 (LWP 19441)]
Spectrum visual data thread started.
[New Thread 0x7fffe8f41700 (LWP 19442)]
Spectrum visual data thread started.
[New Thread 0x7fffd35d7700 (LWP 19443)]
FFT visual data thread started.

Audio Device #0 PulseAudio
        Default Output? Yes
        Default Input? Yes
        Input channels: 2
        Output channels: 2
        Duplex channels: 2
        Native formats:
                16-bit signed integer.
                32-bit signed integer.
                32-bit float normalized between plus/minus 1.0.
        Supported sample rates:
                8000hz
                16000hz
                22050hz
                32000hz
                44100hz
                48000hz
                96000hz

Available vertical sync SwapInterval functions: 
        glxSwapIntervalEXT: No
        DRI2SwapInterval: No
        glxSwapIntervalMESA: No
        glxSwapIntervalSGI: No
No vertical sync swap interval functions available.
Loaded font 'Bitstream Vera Sans Mono' from '/home/user/src/CubicSDR/build/x64/vera_sans_mono16.fnt', parsed 167 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/home/user/src/CubicSDR/build/x64/vera_sans_mono12.fnt', parsed 167 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/home/user/src/CubicSDR/build/x64/vera_sans_mono18.fnt', parsed 167 characters.
[New Thread 0x7fffd19d6700 (LWP 19444)]
SDR enumerator starting.
SoapySDR init..
        API Version: v0.4.0-gd17177f3
        ABI Version: v0.4-2
        Install root: /usr
        Loading modules... 
        Available factories...nullJust 'null' factory found.
[xcb] Unknown request in queue while dequeuing
[xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
[... 61 lines cut ...]
#53 0x00007ffff402534c in ?? () from /usr/lib64/libgtk-x11-2.0.so.0
#54 0x00007ffff389e1f3 in g_closure_invoke () from /usr/lib64/libgobject-2.0.so.0
#55 0x00007ffff38b0aed in ?? () from /usr/lib64/libgobject-2.0.so.0
#56 0x00007ffff38b53f5 in g_signal_emit_valist () from /usr/lib64/libgobject-2.0.so.0
#57 0x00007ffff38b57d4 in g_signal_emit_by_name () from /usr/lib64/libgobject-2.0.so.0
#58 0x00007ffff3e566bf in ?? () from /usr/lib64/libgtk-x11-2.0.so.0
#59 0x00007ffff389e430 in ?? () from /usr/lib64/libgobject-2.0.so.0
#60 0x00007ffff38b4e87 in g_signal_emit_valist () from /usr/lib64/libgobject-2.0.so.0
#61 0x00007ffff38b567a in g_signal_emit () from /usr/lib64/libgobject-2.0.so.0
#62 0x00007ffff4019a13 in gtk_widget_map () from /usr/lib64/libgtk-x11-2.0.so.0
#63 0x00007ffff402548a in ?? () from /usr/lib64/libgtk-x11-2.0.so.0
#64 0x00007ffff389e430 in ?? () from /usr/lib64/libgobject-2.0.so.0
#65 0x00007ffff38b4e87 in g_signal_emit_valist () from /usr/lib64/libgobject-2.0.so.0
#66 0x00007ffff38b567a in g_signal_emit () from /usr/lib64/libgobject-2.0.so.0
#67 0x00007ffff4018d30 in gtk_widget_show () from /usr/lib64/libgtk-x11-2.0.so.0
#68 0x00007ffff3e578d5 in gtk_dialog_run () from /usr/lib64/libgtk-x11-2.0.so.0
#69 0x00007ffff70b92dc in wxMessageDialog::ShowModal() () from /usr/lib64/libwx_gtk2u_core-3.0.so.0
#70 0x000000000042358f in CubicSDR::sdrEnumThreadNotify(SDREnumerator::SDREnumState, std::string) ()
#71 0x000000000045ddc8 in SDREnumerator::enumerate_devices(std::string, bool) ()
#72 0x000000000045e475 in SDREnumerator::run() ()
#73 0x00007ffff4f7b5a0 in ?? () from /usr/lib/gcc/x86_64-pc-linux-gnu/4.9.3/libstdc++.so.6
#74 0x00007ffff479b428 in start_thread () from /lib64/libpthread.so.0
#75 0x00007ffff44e1edd in clone () from /lib64/libc.so.6

```

cjcliffe, 2016-01-14:
@TheOncomingStorm that's perfect thanks; I think I know what's doing that and can make a patch to prevent it.

cjcliffe, 2016-01-14:
@TheOncomingStorm I've committed a fix, it's happened before -- basically a modal dialog was trying to be spawned from a background thread and the window manager said no..

To answer why you got that dialog in the first place is written on the dialog itself; please let me know if it works and then let me know if it was an appropriate dialog for the situation :)

TheOncomingStorm, 2016-01-14:
Yep, that helped tremendously and I was able to fix the situation appropriately. 

Thanks

closed 2016-01-14 by TheOncomingStorm

cjcliffe, 2016-01-15:
Good to hear it works as expected :+1: thanks for helping fix that up.