cjcliffe/CubicSDR #255: Bundled hamlib support

cjcliffe, 2016-01-14:
Have attempted to do this on OSX (static, dynamic, etc. with and without modules) so far without success.  Need to investigate for each platform and figure out how to bundle properly.

hamlib bundle support for:
- [ ] OSX
- [ ] Windows
- [ ] Linux?

neilwillgettoit, 2016-04-30:
I got the most up to date master branch build to work with hamlib enabled, but when I try to select the rig model to control, I do not get anything in that menu to select.

neilwillgettoit, 2016-05-02:
Okay, so my findings to get this working: 

brew install hamlib - doesn't not work 

pulling the latest from https://github.com/N0NB/hamlib and compiling it from scratch, then compiling cubicSDR works!

neilwillgettoit, 2016-05-02:
Control from cubicsdr to a rig is very smooth. But if you try to tune around with the vfo on the rig, something about the way it is polling for rig frequency is causing the rig to do a 'two steps forward, one step back' type jitter. I've been able to replicate this with both a Yaesu FT-857d and an Elecraft KX3. 

I do not get this type of behavior in other applications such as rumlogng or n1mm that do very similar type vfo tracking of the radio and plot frequency on bandmaps to the same radios.

Toontje, 2016-05-02:
No need to install hamlib first. In fact, it only makes things more difficult because you need to load the rig list separately as well. Macports or homebrew installs maclib without rig list.
I installed the latest packaged build on a virgin Mac and all is working fine.
I do notice the jittering you mentioned. I thought this was because of me having two programs controlling the radio (logging program and CubicSDR). I didn't find it a huge problem though.

cjcliffe, 2016-05-02:
@neilwillgettoit I think you're correct; the polling is a bit wonky -- I'm not currently using the hamlib callback functions which I think would smooth that out -- right now it's just using a timer that polls every few seconds.

I'll try implementing the callback and push a test build here to try and see if it resolves that.

cjcliffe, 2016-05-02:
@Toontje excellent; it was actually simpler than I thought to get it bundled -- I just had to build it myself and tell it to not build the rig modules as shared files (the default) -- I've yet to see if I can still load additional externals when it's built that way.

cjcliffe, 2016-05-16:
Windows 32 and 64-bit hamlib bundled builds are now up for 0.1.28; windows was a bit trickier and requires MinGW to build the libs.   

Hamlib should now be bundled in all future releases -- closing this up for now.

closed 2016-05-16 by cjcliffe