cjcliffe/CubicSDR #256: TX - Transmitter support

cjcliffe, 2016-01-14:
With HackRF, Audio, etc. support for devices that can transmit through SoapySDR it's now possible to create the framework for transmitting analog or digital signals.

Ideas for transmission support:
- [ ] Create a new thread and manager for transmitters
- [ ] Implement analog modulations
   - [ ] Allow adding transmitter modems to the waterfall
   - [ ] Allow selection of recording device
   - [ ] Display push-to-talk button when transmitter active
   - [ ] Add option to save push-to-talk recording to audio clipboard for delayed playback
   - [ ] Add "capture to audio clipboard" for demodulators that can be played back by transmitters
   - [ ] Add option to source transmission from .WAV file playback, load wavs to clipboard
- [ ] Implement digital modulations
   - [ ] TODO: digital checklist items

vsboost, 2016-01-14:
Oh wow

Analias, 2016-02-19:
Please consider the following likely use cases for single or multiple devices.  We should handle each case to provide flexibility for multiple radios in the case of split operation or satellite operation.

1)  The SDR receiver and transmitter are integrated and share the same controller.  
2)  The SDR receiver and transmitter are separate devices in the same chassi and may be operated independently of each other in either full or half duplex
3)  The SDR receiver and transmitter are completely  separate pieces of equipment that may or may not be completely independent of each other.
4)  The receiver and transmitter may be a mix of SDR and traditional radios (ex. SDR and a TS-2000)

bdn76, 2016-10-30:
Great and up! Any news?