cjcliffe/CubicSDR #261: Compile error

frspin, 2016-01-23:
Current GIT version don't compile with Ubuntu 14.04 and c++ 4.8 (standard compiler for this Ubuntu version).

/home/spin/Scaricati/ham/CubicSDR/src/sdr/SDREnumerator.cpp: In static member function ‘static void SDREnumerator::removeManual(std::string, std::string)’:
/home/spin/Scaricati/ham/CubicSDR/src/sdr/SDREnumerator.cpp:424:28: error: no matching function for call to ‘std::vector<_SDRManualDef>::erase(std::vector<_SDRManualDef>::const_iterator&)’
             manuals.erase(i);

No problem compiling it with Ubuntu 15.10

Franco Spinelli
IW2DHW

cjcliffe, 2016-01-23:
strange probably a GCC difference -- I have a ubuntu 14 VM here I'll take a look

frspin, 2016-01-23:
Il 23/01/2016 16:51, Charles J. Cliffe ha scritto:
> strange probably a GCC difference -- I have a ubuntu 14 VM here I'll
> take a look

Ubuntu 14.04 have GCC 4.8.2 and Ubuntu 15.10 have 5.2.1

Franco Spinelli
IW2DHW

zonduino, 2016-01-23:
I have the same error on openSUSE Leap 42.1 with c++ 4.8 and GCC 4.8.5
All previous builds were successful. 

Timothy Baker

ryanniehaus, 2016-01-24:
Same here linux mint 17.3 gcc 4.8.4-2

sm0yyq, 2016-01-24:
I have same trouble with v0.1.22-alpha and ubuntu 14.04, .. version 0.1.20-alpha compiles well.
It also throw out lots of warnings under compile time, on both versions.

cjcliffe, 2016-01-24:
Should be fixed up now; had a typo that OSX and Windows don't seem to care about :)

frspin, 2016-01-24:
Il 24/01/2016 16:20, Charles J. Cliffe ha scritto:
> Should be fixed up now; had a typo that OSX and Windows don't seem to
> care about :)
> 

I can confirm: the problem is solved on Ubuntu 14.04

Franco Spinelli
IW2DHW

cjcliffe, 2016-01-24:
@frspin great; thanks for the quick response!

closed 2016-01-24 by cjcliffe