cjcliffe/CubicSDR #262: Compiling SoapySDR for OSX

twissmueller, 2016-01-25:
The build description for OSX in the wiki (https://github.com/cjcliffe/CubicSDR/wiki/Build-OSX) states 

    sudo apt-get install git build-essential automake cmake

This is not native to the Mac, what steps do I need to take? Shall I just follow the steps as given in SoapySDR build steps: https://github.com/pothosware/SoapySDR/wiki/BuildGuide#osx

Any recommendations?

cjcliffe, 2016-01-25:
Hmm, those were definitely steps from the Linux guide that must have copied over during the last updates -- I've made a few quick edits to fix the missing / incorrect info -- let me know if that helps.

twissmueller, 2016-01-26:
Great! Got it compiled on OSX 10.10.5. Did a small change to the documentation by rearranging one part to the top.

closed 2016-01-26 by twissmueller

cjcliffe, 2016-01-27:
good to hear; thanks for fixing that up!