cjcliffe/CubicSDR #263: CubicSDR crashes when opening a NetSDR+

gastonet, 2016-01-25:
Hi Charles et all. Tried v0.1.20 and v0.1.22 with a NetSDR+ from RFSpace and CubicSDR crashes when opens the SDR (I can hear the relays chattering just before the crash).
App works flawlessly with an RTL dongle, also recognise perfectly the NetSDR.

Here's the crashlog:

http://pastebin.com/WsDKzNNj

I'm very available to make any test, so count on me.

Best.

cjcliffe, 2016-01-25:
Interesting, looks like a crash somewhere in SoapySDR during the initial readStream -- I may need to summon @guruofquality to help pinpoint what went wrong.

guruofquality, 2016-01-25:
```
Thread 18 Crashed:
0   libsystem_platform.dylib        0x00007fff98a0801c _platform_memmove$VARIANT$Haswell + 252
1   com.cubicproductions.cubicsdr   0x0000000104204f0d SDRThread::readStream(ThreadQueue<SDRThreadIQData*, std::__1::list<SDRThreadIQData*, std::__1::allocator<SDRThreadIQData*> > >*) + 157
2   com.cubicproductions.cubicsdr   0x0000000104205073 SDRThread::readLoop() + 99
3   com.cubicproductions.cubicsdr   0x0000000104205cb9 SDRThread::run() + 393
4   com.cubicproductions.cubicsdr   0x0000000104176d64 IOThread::threadMain() + 20
5   com.cubicproductions.cubicsdr   0x0000000104161b52 void* std::__1::__thread_proxy<std::__1::tuple<void* (IOThread::*)(), SDRThread*> >(void*) + 114
6   libsystem_pthread.dylib         0x00007fff99eabc13 _pthread_body + 131
7   libsystem_pthread.dylib         0x00007fff99eabb90 _pthread_start + 168
8   libsystem_pthread.dylib         0x00007fff99ea9375 thread_start + 13
```

It looks like its crashing after SoapySDR::readStream in the SDRThread::readStream. Perhaps the RF space stream implementation returned some erroneous number of samples read (too large perhaps)?

This might be annoying, but we need a few prints to see what readStream() is doing.

* One in SDRThread::readStream() to see how much SoapySDR::readStream returned
* In here, this is where SoapyOsmo (which supports rf space) would call read into the rf space stream implementation: https://github.com/pothosware/SoapyOsmo/blob/master/GrOsmoSDRInterface.hpp#L176
* And here where rfspace actually pulls samples from the socket, there is a lot going on here, certainly an opportunity for some edge cases to get exercised: https://github.com/pothosware/SoapyOsmo/blob/master/gr-osmosdr/lib/rfspace/rfspace_source_c.cc#L779

cjcliffe, 2016-01-25:
@guruofquality thanks, I'll make up a branch with some additional logging to help debug this

gastonet, 2016-01-25:
Thanks both! How can I help? I can put a Mac and the NetSDR+ with TeamViewer or something if anyone needs to troubleshoot. 
Just let me know. 
Thanks a lot.

cjcliffe, 2016-01-26:
@gastonet are you building from source?  If not I can make a debug logging binary package when ready to test -- basically you'll just have to run it from Terminal and copy the output.

gastonet, 2016-01-26:
No @cjcliffe I've downloaded the binaries right from GitHub. Go ahead, make the package and I'll run the tests.

cjcliffe, 2016-01-26:
@gastonet here's the test binary package (note I zipped the DMG so GitHub would accept): 

[CubicSDR-0.1.22-Darwin-netsdr_debug.dmg.zip](https://github.com/cjcliffe/CubicSDR/files/104292/CubicSDR-0.1.22-Darwin-netsdr_debug.dmg.zip)

Just mount the image and run in a fresh Terminal:

```
/Volumes/CubicSDR-0.1.22-Darwin/CubicSDR.app/Contents/MacOS/CubicSDR
```

Any copy the resulting output into another pastebin or Gist -- https://gist.github.com/

Thanks!

gastonet, 2016-01-26:
Hi again @cjcliffe here's the output! And thank you for your time!

https://gist.github.com/gastonet/fa9982af49c447da12aa

cjcliffe, 2016-01-26:
@gastonet lol, yeah this is definitely my bad; I made no account for SoapySDR returning *more* samples than I requested, @guruofquality got it right on the first guess :)

Looks like it always returns 256; my code checks for less or equal to but not greater than what was requested..  I think I'll need some sort simple of overflow buffer for the next read.  I'll fix that up and make you another test build -- if that works I'll update the current release builds.

gastonet, 2016-01-26:
Yay! Good to see you are on the right way! Let me know when the another test build is ready so I can test again.

cjcliffe, 2016-01-26:
@gastonet hopefully this should do the trick; doubled the internal buffer and added an overflow buffer that will supplement subsequent reads as needed.

Here goes: 
[CubicSDR-0.1.22-Darwin-netsdr-fixtest.dmg.zip](https://github.com/cjcliffe/CubicSDR/files/104343/CubicSDR-0.1.22-Darwin-netsdr-fixtest.dmg.zip)

guruofquality, 2016-01-26:
> 
Requesting 56 elements 
Read 256 elements 

Lol.

Here is the rfspace snippet at fault, from the work() function. Oops no bounds checking!

```
  size_t rx_samples = (rx_bytes - HEADER_SIZE - SEQNUM_SIZE) / (sizeof(int16_t) * 2);
  //so if rx_samples is > noutput_items, it will overwrite buffers here...

  //and then here is where the returned value becomes larger...
  noutput_items = rx_samples;

  return noutput_items;
```

It probably needs to store the remainder and copy it into the output on the next call(s). I don't know who to call about it. But its a bug in rfspace source for sure.

gastonet, 2016-01-26:
Hi @cjcliffe sorry about the delay (I had to be father for a while). Unfortunately isn't working yet, here's the new log:
https://gist.github.com/gastonet/bb1ca413e49fb5c5afc7

cjcliffe, 2016-01-26:
@gastonet those particular log messages should be gone now; I think you might still have the first DMG I sent mounted or else I re-packed the same one :)    Try un-mounting all the CubicSDR DMG's and just mount the netsdr-fixtest one and see if that does it.

cjcliffe, 2016-01-26:
@guruofquality am I going to end in unknown memory on those < 256 request packets?  Not quite sure if I read that right.

gastonet, 2016-01-26:
@cjcliffe I'm pretty sure I'm running the right binary from the right package, I've deleted all the old packages and re-downloaded the "fixtest.dmg". Also please have a look to the log, It took me 3 or 4 times clicking on the refresh button because CubicSDR was unable to find the NetSDR+
Here's the recent log:
https://gist.github.com/gastonet/5a2a402a6cb789c8608f

guruofquality, 2016-01-26:
@cjcliffe In short, the rfspace has a bug where it cant handle input buffers smaller than the transfer size. In this case transfer size appears to be 256 samples. If  the buffer is smaller than 256 samples, the rfspace implementation will overrun the buffer. You might have to work around it by ensuring that a minimum amount of space is available in the readStream() buffer before making a call to work around this. Or alternatively, patch the rfspace work() function.

Something that might feel less hacky for the workaround idea: Soapy SDR has a getStreamMTU() call, which equates to the buffer size used in receive transfers. If you ensured that the readStream() buffer size was always >= to the MTU, then 1) it might result in less transfer calls overall because fragments will be avoided. And 2) in this case the rfspace transfer size is always less than the default stream MTU (this is the hacky assumption part).

cjcliffe, 2016-01-27:
@gastonet yup, I see now your log just had both versions run in it; the one I was looking for was further down -- I'll do another debug logging build for you and see if we can spot the failure.  I'll also try simulating some "bad" inputs here to make sure it's doing what it's supposed to do.

gastonet, 2016-01-27:
Great then @cjcliffe I'll wait for the new build. I'll be a couple of more hours around so count on me for any tests. And thanks for the Command-K I'll use it instead of clear command.
Best.

cjcliffe, 2016-01-27:
@gastonet whoops -- latest crash was due to some bad pointer arithmetic on my part :)

Here's the latest attempt:

[CubicSDR-0.1.22-Darwin-netsdr-fixtest2.dmg.zip](https://github.com/cjcliffe/CubicSDR/files/105924/CubicSDR-0.1.22-Darwin-netsdr-fixtest2.dmg.zip)

gastonet, 2016-01-27:
Wee now it doesn't crash anymore! but... can't tune, although I can see the center frequency I can't tune any frequency, in fact, no frequency is shown where's supposed to be (I know where it is because it works with the RTL dongle).
Here's the log, also It took me just 2 clicks on the refresh button to be able to see the NetSDR.
https://gist.github.com/gastonet/74c5c896e91bfa8462b4

cjcliffe, 2016-01-27:
@gastonet that's.. interesting -- let me try adding @guruofquality 's getStreamMTU() suggestion and make overflowing the norm in favour of device MTU and cook another build -- and if that doesn't do the trick maybe we'll take a trip to rfspace code to make those fixes first.

cjcliffe, 2016-01-27:
@gastonet here's another attempt; CubicSDR should always request the stream MTU from SoapySDR now and handles the difference itself:

[CubicSDR-0.1.22-Darwin-netsdr-fixtest3.dmg.zip](https://github.com/cjcliffe/CubicSDR/files/105963/CubicSDR-0.1.22-Darwin-netsdr-fixtest3.dmg.zip)

gastonet, 2016-01-27:
Ok @cjcliffe here's the report. I've tried to run it two times. Both times CubicSDR found the NetSDR+ right on the first attempt, which is really good. Now, on the other hand, both times crashed trying to open NetSDR. There are two different errors as far I can see.
First crash: https://gist.github.com/gastonet/8bb4c2b13db59194c27f
Second crash: https://gist.github.com/gastonet/5c686205baaad269fa73

cjcliffe, 2016-01-27:
@gastonet Ok, I'm going to crank the CubicSDR stream logging up to 11 and we'll see what that reveals.

cjcliffe, 2016-01-27:
Forcing MTU to 1024 here causing some problems; looks like it's still something I need to fix :)

gastonet, 2016-01-27:
Go ahead @cjcliffe I'm here.

cjcliffe, 2016-01-27:
@gastonet Ok, that was just a typo between the mtu and element buffer count (over-wrote some memory causing random errors);  Here's another attempt:

[CubicSDR-0.1.22-Darwin-netsdr-fixtest4.dmg.zip](https://github.com/cjcliffe/CubicSDR/files/105977/CubicSDR-0.1.22-Darwin-netsdr-fixtest4.dmg.zip)

gastonet, 2016-01-27:
ok @cjcliffe here's what happened. I've tried two times again. Both took between 5 and 8 attempts to found the NetSDR. Both times I was able to tune one FM broadcast but at some point (maybe doing fine tuning) CubicSDR was freezed, not crashed, not hang, no beachball. The sound stopped and the waterfall and other windows was like paused. Also I had to use Control-C to kill the process.
Here are two logs:
https://gist.github.com/gastonet/c2c0b750937711cd1294
https://gist.github.com/gastonet/67654ada85645e4c1bfd

cjcliffe, 2016-01-27:
@gastonet ok, I think I've taken this about as far as I can on the CubicSDR side, I think it makes the most sense to fix the rfspace code to rule that out as the remaining factor.   

I'm going to prepare a pull request and do some review to make sure I don't see anything in CubicSDR left to patch and perhaps @guruofquality can provide some guidance on where we go from here.

gastonet, 2016-01-27:
@cjcliffe ok many thanks for your incredible effort. I'll be open to any test you and @guruofquality needs.

cjcliffe, 2016-01-27:
@gastonet thanks for being on-hand for quick turnaround on testing -- I'll let you know as soon as there's more to test.

Cheers!

gastonet, 2016-01-27:
@cjcliffe glad to being helpful on such a great software.
Best!

cjcliffe, 2016-04-01:
@gastonet thanks for your help on this; going to close it up for now -- please report any additional issues if they arise.

closed 2016-04-01 by cjcliffe

gastonet, 2016-04-01:
Thank you @cjcliffe you made an awesome work with CubicSDR and now I'm happy to be able to enjoy it using my NetSDR+
One more thing probably useful before 0.26 release: The version I'm currently using does not recognize the RTL dongles. Best!

cjcliffe, 2016-04-01:
@gastonet the build I made for you was NetSDR+ only to keep building simple :)   Next binary release will have the full set again.