cjcliffe/CubicSDR #272: Improvements

frspin, 2016-01-30:
Two improvements to CubicSDR for using in HAM bands
- a NBFM button with a preset of 12500 Hz as bandwidth
- save of Automatic/Manual gain control for each device (required for SDRPlay on HF)

Regards

Franco Spinelli
IW2DHW

cjcliffe, 2016-01-31:
Thanks @frspin, sounds good -- added to the list for enhancements.

jdeitch, 2016-02-11:
Can I add a request to this list?  How about a STOP button?  Sometimes you may want to halt the player but not exit.  Skype, etc. come to mind.

Jim

cjcliffe, 2016-02-11:
@jdeitch the "Refresh" button on the SDR Devices dialog is technically also a "Stop" button.   I can see that adding a Start/Stop to the menu might be a good idea as well -- it's simple enough I'll try and see.

cjcliffe, 2016-05-12:
NBFM, Start/Stop, saving AGC state / gains per-device and NBFM are implemented and now merged to master.

Cheers!

closed 2016-05-12 by cjcliffe