cjcliffe/CubicSDR #274: rtl-sdr not list in device

mohammadninja, 2016-02-01:
hi 

when i open the cubic sdr 
in the cubic sdr device is empty 
[image]
[image]

cjcliffe, 2016-02-01:
@mohammadninja it's possible you need to blacklist the RTL-SDR device as described here: https://opendesignengine.net/news/53 

It's also possible you may not have SoapyRTLSDR installed; can you paste the output from running:

```
SoapySDRUtil --info
```

and 

```
SoapySDRUtil --find
```

Also I see a problem with the dialog in the screenshot you've posted which I'll investigate.

mohammadninja, 2016-02-01:
yes the textbox for remote address is not working

i use windows and when i use sdrsharp rtl sdr is working good

i not found SoapySDRUtil  binary for windows

mohammadninja, 2016-02-01:
`C:\Program Files\PothosSDR\bin>SoapySDRUtil --info
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

API Version: v0.4.1-PothosSDR-2016.01.29-vc14-x64
ABI Version: v0.4-2
Install root: C:\Program Files\PothosSDR
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules/airspySupport.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules/audioSupport.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules/bladeRFSupport.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules/HackRFSupport.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules/miriSupport.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules/osmosdrSupport.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules/RedPitaya.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules/remoteSupport.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules/rtlsdrSupport.dll
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules/uhdSupport.dll
Loading modules... Win32; Microsoft Visual C++ version 14.0; Boost_106000; UHD_003.009.002-0-gf18abe54

done
Available factories...airspy, audio, bladerf, hackrf, miri, null, osmosdr, redpitaya, remote, rtlsdr, uhd,
`
`C:\Program Files\PothosSDR\bin>SoapySDRUtil --find
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Win32; Microsoft Visual C++ version 14.0; Boost_106000; UHD_003.009.002-0-gf18abe54

Found Rafael Micro R820T tuner
`

cjcliffe, 2016-02-01:
@mohammadninja Ok -- it's likely that I need to update the libraries I'm building against for SoapySDR.

I see you have pothos installed which will conflict if the bundled SoapySDR doesn't match up with the modules -- CubicSDR always tries to use your locally installed Soapy modules first if available.

I'd like to integrate with the pothos releases at some point too so that we're synchronized but for now there's an extra flag you can use with CubicSDR to enforce using bundled modules only:

```
CubicSDR -b
```

or

```
CubicSDR --bundled
```

You should be able to edit the .lnk file for CubicSDR and add that as a command line parameter so it runs automatically.

Optionally removing the SoapySDR DLLs and modules from the CubicSDR application directory *may* allow your locally installed version to work.  I should possibly make them optional in future installers..

cjcliffe, 2016-02-04:
Layout for the add device box should be fixed in master; still have to test here before closing it up.

cjcliffe, 2016-02-07:
Now fixed and tested in master; not perfect but will take another look when I rework the dialog.

closed 2016-02-07 by cjcliffe