cjcliffe/CubicSDR #276: Adding CubicSDR build to PothosSDR

cjcliffe, 2016-02-03:
- Add any required files or build options for integrating with PothosSDR release builds

guruofquality, 2016-02-04:
Little intro
====

First off, the build environment is totally open source and anyone can replicate it (a brave few have). If you have the patience to download and unzip/install a bunch of packages, it might be worth it. Also, you could comment out some of the software selections in the top level CMakeLists to save some effort as well.

* Super helpful [build guide](https://github.com/pothosware/PothosSDR/blob/master/README.build)  (thanks @bkerler)
* And [manifest](https://github.com/pothosware/PothosSDR/blob/master/README.md#prebuilt-binaries-manifest) for external downloads

Extending for cubicsdr
====

Basically extending the build to CubicSDR is a matter creating another cmake file in the top level (like BuildCubicSDR.cmake) directory with several ExternalProject_Add() commands for CubicSDR and dependencies that we will build.

Prebuilt dependencies are [listed here](https://github.com/pothosware/PothosSDR/blob/master/InstallPrebuilt.cmake) (although it could be organized into a different file, just an example). And they typically have an install command to copy the required runtime dlls over.

Cubicsdr dependencies
====

We need to figure out which dependencies will be:
* downloaded as "prebuilt"
* build by the environment
* or checked into [external/](https://github.com/cjcliffe/CubicSDR/tree/master/external)

I would say that anything that is large and hard to build that has reputable installers for should be treated as prebuilt (this is probably the case for wx for example). Packages that can easily be built from source with git/cmake/MSVC should be added to the build with ExternalProject_Add(). -- and if its not quite a cmake project, ExternalProject_Add() is pretty flexible FWIW... And then anything else that doesnt fit that paradigm should just be used from external/, which I guess is easiest.

Current dependencies that are taken care of by the environment:

* FFTW (prebuilt - see manifest)
* SoapySDR (built in environment)

Other dependencies. We should figure out prebuilt/or not, which version, url, or branch:

* hamlib
* liquid
* wx sdk
* OpenGL
* any others?

guruofquality, 2016-02-05:
Anyway, let me know what you have been obtaining the various dependencies so far (including the ones checked into external/). I will try to either download them or get them into the build system... And I will see what happens.

cjcliffe, 2016-02-06:
@guruofquality thanks I haven't had a chance this week to dig into this but I'm around this weekend.

Basically outside of CubicSDR's own external/ folder the only dependencies are:

- CMake
- FFTW (fftw3f / fftw3-single)
- liquid-dsp
- SoapySDR
- wxWidgets
- OpenGL

Everything else should be built-in or was used as a tool and is referenced in the README mostly as an acknowledgement.

If you can get hamlib to distribute properly please do, I've been having problems getting it to find modules after bundling.

guruofquality, 2016-02-06:
* **branch** Working through this, I got started on this branch: https://github.com/pothosware/PothosSDR/tree/cubicsdr_work

* **wx binaries** I found prebuilt wx widgets for msvc 2012 and 2013, so I am starting with 2013: http://sourceforge.net/projects/wxwindows/files/3.0.2/ I might have to build for 2015 in this case.

* **first error** I am building off the master branch and I got this error: 
```
  B:\PothosSDR_vc12\Source\CubicSDR\src\sdr\SDRDeviceInfo.cpp(5): error C2664: 'std::atomic_bool::atomic_bool(const std::atomic_bool &)' : cannot convert argument 1 from
 'bool' to 'const std::atomic_bool &' [B:\PothosSDR_vc12\Build\CubicSDR\CubicSDR.vcxproj] [B:\PothosSDR_vc12\CubicSDR.vcxproj]
```

* **specify fftw** It would be nice to manually specify the location of dependencies. Right now I have fftw already provided so I would like to be able to pass these in externally. Perhaps they will only be set to the copy in external/ when the variable is not already set. Ex ```cmake -DFFTW_INCLUDES=...```

```
	SET(FFTW_INCLUDES "${PROJECT_SOURCE_DIR}/external/fftw-3.3.4/")
	SET(FFTW_LIBRARIES "${PROJECT_SOURCE_DIR}/external/fftw-3.3.4/${EX_PLATFORM}/libfftw3f-3.lib")

#change this to... etc

if (NOT DEFINED FFTW_INCLUDES)
    SET(FFTW_INCLUDES "${PROJECT_SOURCE_DIR}/external/fftw-3.3.4/")
endif()
if (NOT DEFINED ...

```

* **build liquid** I would like to build liquid in the environment as well if possible. I honestly had no idea it could be built with MSVC. Is this a special branch or custom VC project or fork? Any tips for this one would be appreciated.

cjcliffe, 2016-02-06:
first error - I've added a patch, I've only been using MSVC2015 here so I missed that one

second part - I've added CACHE STRING and description to FFTW and LIQUID variables which should allow you to set them externally

build liquid - TBH http://liquidsdr.org/ circa late 2014 used to contain a .ZIP download for liquid-dsp 1.2.0 which contained the DLL that's still used today -- and I don't know how it was built..  The external folder for liquid-dsp contains the contents of that zip.

I've got a solution that requires cygwin to build FFTW+Liquid-DSP together plus including some cygwin DLL's but I'm reluctant to make that complication the default when the original build is still working fine.  That said I'm having problems now using newer features (with latest LSB/USB modem updates, IIR lowpass filter is new and I had to #define fork WIN32 and use FIR) and would very much like to get liquid-dsp building an MSVC compatible DLL.

guruofquality, 2016-02-06:
Thanks for the changes. It also looks like there aren't any install commands because I didnt see an install target generated. Since the installer packages the install directory, a ```install(TARGET CubicSDR DESTINATION bin) ``` would be useful here (vs creating a post install copy command to where ever the binary happens to be).

Sounds like a mess! Didnt know that. I have some interest in getting liquid building in cmake + msvc for some eventual modem demos with pothos. Maybe I will give it a shot. I got gnuradio building in cmake + msvc, so it can't be that bad. ;-)

guruofquality, 2016-02-06:
Minor findwxwidgets bug. I need to set an extra variable (```WX_ROOT_DIR```) to trick the find script, or else it forgets the ```wxWidgets_LIB_DIR``` setting until a second configure. I cant have it failing configure, so this was my work-around. Not to bad:

```
cmake ...etc...
        -DWX_ROOT_DIR=${wxWidgets_ROOT_DIR}
        -DwxWidgets_ROOT_DIR=${wxWidgets_ROOT_DIR}
        -DwxWidgets_LIB_DIR=${wxWidgets_LIB_DIR}
```

This snippet from FindwxWidgets causes the problem: 

```
  # If wxWidgets_ROOT_DIR changed, clear lib dir.
  if(NOT WX_ROOT_DIR STREQUAL wxWidgets_ROOT_DIR)
    set(WX_ROOT_DIR ${wxWidgets_ROOT_DIR}
        CACHE INTERNAL "wxWidgets_ROOT_DIR")
    set(wxWidgets_LIB_DIR "wxWidgets_LIB_DIR-NOTFOUND"
        CACHE PATH "Cleared." FORCE)
  endif()
```

Is there an advantage to the [checked in findwxwidgets](https://github.com/cjcliffe/CubicSDR/blob/master/cmake/Modules/FindwxWidgets.cmake) vs the [cmake provided one](https://cmake.org/cmake/help/v3.0/module/FindwxWidgets.html)?  Anyway, just FYI if you had a similar problem.

guruofquality, 2016-02-06:
Got it built. Looks like it crashes in ```void GLPanel::calcTransform(mat4 transform_in) {``` because it seems that the GLPanel is null. Perhaps its because this is a VM without gl and whatever factory returned the GLPanel returned a nullptr. Anyway, if thats the case, it may be something worth catching/checking for null before it segfaults. I will try on a real machine tomorrow...

cjcliffe, 2016-02-06:
@guruofquality yeah the included wxWidgets find script may no longer be relevant; I think it was an issue with an late version of CMake 2.x that needed it.

Not sure why the GLPanel itself would be null but I'm guessing that's probably true if the GL canvas didn't initialize.

guruofquality, 2016-02-07:
> yeah the included wxWidgets find script may no longer be relevant; I think it was an issue with an late version of CMake 2.x that needed it.

You might want to conditionally use it if the cmake version is new enough (in case they patched something). Anyway, I will keep the work-around in, it seems harmless enough. Just FYI if if was giving you an issue as well.

> Not sure why the GLPanel itself would be null but I'm guessing that's probably true if the GL canvas didn't initialize.

I didn't have any luck on another machine. The wx widgets seems to render right before the crash, so I don't think its an issue with the pre-built wx or ABI mismatch. I will try manually building wx for vc14 which I have to do anyway.

I'm not really sure. Any "gotchas" that you might know about? Like a missing opengl dll to install... I feel like I missed something obvious. Install directX or something?

> There's an install / uninstall target for Linux. For OSX it creates an .app bundle and for windows NSIS creates the installer -- both through CPack but you have to enable the relevant bundle/build flags.

I guess what I mean is that there is no equivalent of  ```IF (WIN32 AND **NOT** BUILD_INSTALLER)```. There is no regular install option for windows. The other OSs seems to have a case for regular install and install for packaging.

For my purposes, I'm not using the CubicSDR NSIS packaging stuff. I'm just doing stock builds of tons various packages, and NSIS packaging the resulting directory (nice start menu shortcuts too of course).

I basically need to fix this by either adding install commands to CubicSDR CMakeLists such that CubicSDR.exe and libliquid.dll (others?) end up in ```DESTINATION bin```. Or doing the equivalent in the PothosSDR CMakeLists (which is a little more hodge podgy).

cjcliffe, 2016-02-07:
@guruofquality ok, I wasn't aware there was a native install target for Windows; I've always used NSIS,  WiX or InstallShield in the past.  I think it's going to be fairly similar to the build_installer section for Windows either way.  I'll peek at what Pothos is doing and try to line it up.

cjcliffe, 2016-02-07:
@guruofquality can't think of any Gotcha's except that I'm only building with MSVC15 now; I haven't tried anything older since I started building with SoapySDR in Windows.   Previous RTL-SDR only builds I was doing all 64-bit builds with TDM-GCC64 and 32-bit used MSVC -- a bit of stuff related to that can still be found in the CMakeLists.

guruofquality, 2016-02-07:
I followed your wx instructions for vc14. Much better results. Its notable that the build results in static libs for wx, where I was previously using prebuilt dynamic libs+dlls. I should give a vc12 build a try to confirm. Also the font graphics are missing (doh!):

[image]

> @guruofquality ok, I wasn't aware there was a native install target for Windows;

If there are cmake install commands, you get an install target. So ```cmake --build dir --config Release --target install``` will install the files into the CMAKE_INSTALL_PREFIX.

As a rule of thumb, basically I just put dlls and exes into "bin"

> Do my binary releases for Win64 or Win32 work for you?

Good idea. I will try this next

guruofquality, 2016-02-10:
Little update:

* The home-made wx for vc12 behaves identically to the prebuilt (it crashes). I will probably be supporting vc12 for a while, so if we ever figure this one out I can package it. I guess for now I will skip cubic sdr for vc12 though.
* The vc14 binary that I built works fine as far as i can tell except the graphics text is missing (see screenshot previous post)
* The prebuilt vc14 installer for cubicsdr has working graphics text. I built wx exactly the same way as the wiki suggested. So maybe your CubicSDR build has different flags set related to unicode or something... Can you share your working CubicSDR CMakeCache.txt for comparison?

cjcliffe, 2016-02-11:
I'll boot up my windows dev environment shortly and get you my release build cache for win64 and win32.  The UI text is 100% OpenGL and loaded from a texture file; if it's not loading it's because the font .PNG files or BMFont definitions are missing from the binary folder -- it looks in the working directory first; and then moves on to the binary directory for resource search.  On Linux you can specify an additional resource search location at build time.

guruofquality, 2016-02-14:
OK, that was it. The font resources needed to be installed.

So I replicated the install rules in the top level build like so:

```
ExternalProject_Get_Property(CubicSDR SOURCE_DIR)
ExternalProject_Get_Property(CubicSDR BINARY_DIR)
file(GLOB CUBICSDR_RESOURCES "${SOURCE_DIR}/font/*.*")
install(
    FILES
        ${SOURCE_DIR}/external/liquid-dsp/msvc/64/libliquid.dll
        ${BINARY_DIR}/x64/${CMAKE_BUILD_TYPE}/CubicSDR.exe
        ${CUBICSDR_RESOURCES}
    DESTINATION bin
)
```

So thats a workaround, but my install target wishlist for CubicSDR would look like the following addition to the CubicSDR/CMakeLists.txt:

```
IF (WIN32 AND NOT BUILD_INSTALLER)

    ADD_DEFINITIONS(
	    -DRES_FOLDER="../share/cubicsdr/"
	)	

    INSTALL(TARGETS CubicSDR DESTINATION bin)
        install(FILES 
            ${CUBICSDR_RESOURCES}
            ${PROJECT_SOURCE_DIR}/src/CubicSDR.png
    DESTINATION share/cubicsdr)
```

I'm not sure if the RES_FOLDER relative path would work, but the idea was to support a relocatable install and to also put the icons into the share directory (and not bin).

guruofquality, 2016-02-19:
@cjcliffe: As it stands now CubicSDR will be in the next installer for MSVC 2015 and 2013. However it currently crashes on 2013. Should we open a bug on that or just let it be?

Any resolution on the install target? I would like to give you a patch for a windows install target, but it not I will keep the external install rules and putting the resource files in bin/

And somewhat related, I did make a working [CMakeLists for liquid](https://github.com/pothosware/liquid-dsp/blob/master/CMakeLists.txt). I started tweaking it for MSVC, but it seems that a lot of the code revolves around c99 "complex float" which still isnt supported even in 2015. I don't see building it in MSVC without substantial code changes :-(

cjcliffe, 2016-02-20:
@guruofquality I'll take a look at patching that now; I'm not overly concerned with supporting 2013 as it has always been problematic -- I'm pretty sure you need to swap out the included DLL's with earlier repo versions for it to even work; and at that point you'll need to fix it to compile the 64-bit version with TDM-GCC to work with those DLLs which gets messy.

As a note for liquid-dsp I don't use the complex float option; I'm always working with liquid_float_complex with single precision and I don't use complex float in CubicSDR either.

guruofquality, 2016-02-20:
>  I'm not overly concerned with supporting 2013 as it has always been problematic

I thought that it might be useful for someone that needed to develop with 2013, but wanted to have the full suite of tools. Its more of an edge case. Knowing that I will just disable it for the 2013 build entirely.

> I'm pretty sure you need to swap out the included DLL's with earlier repo versions for it to even work;

Perhaps my crash was related to C++ DLLs for MSVC 2015 being ABI incompatible. I'm sure the C libraries were fine, but the others may not have been...

> As a note for liquid-dsp I don't use the complex float option; I'm always working with liquid_float_complex with single precision and I don't use complex float in CubicSDR either.

So the liquid.h header has a typedef liquid_float_complex thats fairly cross platform. Its just that internally liquid uses the c99 complex type. So compiling against liquid is easier than compiling liquid itself.

cjcliffe, 2016-02-20:
On a side note it's possible (I've got a working copy here) to compile liquid-dsp and fftw3 optional dependency together using cygwin/TDMGCC and link the resulting files against MSVC2015; it just requires including some DLLs from TDMGCC and cygwin.

cjcliffe, 2016-02-20:
@guruofquality intall target seems to work; should I add libliquid.dll / fftw dll to the bin folder as well or are you handling that somewhere already?

cjcliffe, 2016-02-20:
@guruofquality install target patch is in now; pretty much a copy paste of what you provided -- please feel free to make a pull request with anything that helps out with the pothos build; I'm a bit busy this weekend but I can definitely click a few merge buttons ;)

guruofquality, 2016-02-20:
@cjcliffe sounds good. I will give the install target and the resource path a quick test with the complete installer this weekend

guruofquality, 2016-02-20:
> intall target seems to work; should I add libliquid.dll / fftw dll to the bin folder as well or are you handling that somewhere already?

@cjcliffe If its worth the trouble, ideally the install target would install these if they were from the internal copy. Right now I have a rule for installing liquid from cubic source external/ and fftw is part of the environment already. So I am happy with that as it is.

guruofquality, 2016-02-22:
@cjcliffe one last annoying thing. The relative path for the resources is only good when cubicsdr.exe is in the CWD (like from command line) or in the "start in" path for a shortcut. I would like to propose a function to get the absolute path to the executable to robustify finding the resources. The following patch works, but it may not be the cleanest way to do this. Thoughts?

```
diff --git a/src/util/GLFont.cpp b/src/util/GLFont.cpp
index 76d1cb1..33cea52 100644
--- a/src/util/GLFont.cpp
+++ b/src/util/GLFont.cpp
@@ -4,6 +4,25 @@
 #include <fstream>
 #include <algorithm>
 
+#ifdef _MSC_VER
+#include <windows.h>
+
+static std::string getExePath(void)
+{
+    HMODULE hModule = GetModuleHandle(NULL);
+    char path[MAX_PATH];
+    GetModuleFileNameA(hModule, path, MAX_PATH);
+
+    char drive[_MAX_DRIVE];
+    char dir[_MAX_DIR];
+    char fname[_MAX_FNAME];
+    char ext[_MAX_EXT];
+    _splitpath_s(path, drive, dir, fname, ext);
+
+    return std::string(drive) + std::string(dir);
+}
+#endif
+
 #ifndef RES_FOLDER
 #define RES_FOLDER ""
 #endif
@@ -158,6 +177,9 @@ std::string GLFont::getParamValue(std::string param_str) {
 void GLFont::loadFont(std::string fontFile) {
     
     std::string resourceFolder = RES_FOLDER;
+    #ifdef _MSC_VER
+    resourceFolder = getExePath() + "/" + resourceFolder;
+    #endif
 
     wxFileName fontFileName = wxFileName(resourceFolder, fontFile);
     
```

cjcliffe, 2016-02-22:
@guruofquality seems reasonable to me; I've applied the patch as provided -- This is basically the same as what it does with OSX to find the .app bundle path and point it to the internal resources folder.

cjcliffe, 2016-03-04:
Install target now includes the fftw/liquid DLLs, closing this one for now; feel free to open a new issue if I can help improve anything.

closed 2016-03-04 by cjcliffe

guruofquality, 2016-03-04:
No problem. I was going to close it when I build the new installer anyway.

> Install target now includes the fftw/liquid DLLs, closing this one for now; feel free to open a new issue if I can help improve anything.

Supposing that FFTW_LIBRARIES or LIQUID_LIBRARIES was supplied externally. That would be an issue, no?

cjcliffe, 2016-03-07:
@guruofquality Good point, I've added FFTW_DLL and LIQUID_DLL that can now be supplied externally and defaults to the usuals in external/

guruofquality, 2016-03-10:
Thanks for all of the support, new installers with CubicSDR are available: https://groups.google.com/forum/#!topic/pothos-users/9TkOlyo608k

cjcliffe, 2016-03-10:
@guruofquality thanks! posted a related tweet as well: https://twitter.com/ccliffe/status/707750210655559681 :+1:

cjcliffe, 2016-03-10:
@guruofquality just trying out the latest build here on windows 10; I did notice that the SDRPlay support DLL module ABI version doesn't match:

```
Probe device driver=sdrplay
←[1m←[31m[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/mo
dules/sdrPlaySupport.dll)
  sdrplay failed ABI check: Library ABI=0.4-2, Module ABI=0.3-0←[0m
Win32; Microsoft Visual C++ version 14.0; Boost_106000; UHD_003.009.002-0-gf18ab
e54

Error probing device: SoapySDR::Device::make() no match
```

Copying the latest module from CubicSDR builds fixes the problem; but might want to check other modules in case they didn't get updated or something.

Cheers!

guruofquality, 2016-03-10:
I'm not shipping any SDRPlay modules in the installer. Wherever that SDRPlay module came from, its quite out of date. Maybe a remnant on the system?

guruofquality, 2016-03-10:
Oh one more note. I linked to http://cubicsdr.com/ in a few places on the wiki here: https://github.com/pothosware/PothosSDR/wiki but if there is a preferable like an instruction page, or windows specific setup, or we can make a PothosSDR/wiki/CubicSDR with some help app notes -- let me know.

I did something similar for GQRX which basically says to run the tutorial steps to check if the device is detected. I was thinking something along those lines, maybe a helpful screenshot, and perhaps SDRPlay specific post install instructions.