cjcliffe/CubicSDR #278: High CPU usage/clock speed

jacobseptember, 2016-02-07:
Following in the same vein as issue 150.

On my variable-clock-speed Intel E3-1271v3 (four cores w/ HT), CubicSDR seems to force the clock speed all the way up to the boost speed:

[image]

It does not seem to use a full CPU core, unless I'm misunderstanding how Activity Monitor measures usage—runs around 50% minimized to Dock, 80% if visible on-screen.

I'm running the latest OS X compiled binary (0.1.20), but I can do profiling in Xcode if you tell me what you need.

jacobseptember, 2016-02-07:
This problem is not apparent in compiled binary 0.1.15:
[image]

In addition—I may just be slow, but I can't seem to figure out how to toggle stereo output in the latest version.

dc1rdb, 2016-02-07:
@jacobseptember: in the current version there are separate buttons for FM mono (FM) and FM stereo (FMS) in the upper left part of the window.

jacobseptember, 2016-02-07:
Thanks much, @dc1rdb.

cjcliffe, 2016-02-07:
@jacobseptember I think this is still mostly graphics related; I did un-cap the 30fps limit on OSX which I think is causing a significant increase in CPU usage when focused (I notice about 30-35% usage when visible and completely idle).

After 0.2.0 beta release I'll be working towards OpenGL 3.x support which should eliminate most of the CPU related graphics bottlenecks and memory thrashing.

With that said I'll still do some profiling here and see if I can spot the specific reason for the increased CPU usage, there's always a chance there's something new going on :)

cjcliffe, 2016-03-08:
Adding this to the 0.2.0 milestone, going to do some profiling and optimization before release; I'm noticing just the idle GUI has creeped up to around 20% CPU alone here.

cjcliffe, 2016-03-15:
PR #313 is working on addressing some of these issues; if you're able to pull the optimizations_and_fixes branch and build to test locally that'd be good -- if not I can post some binaries in the issue here for testing when ready.

jacobseptember, 2016-03-16:
Very busy this week, but I'll let you know when I give it a whirl. Thanks!

jacobseptember, 2016-03-17:
I tried to build on my laptop (OS X 10.11.0). Took a while to get all the dependencies sorted out, but it ended up working.

Well, the build worked, at least. Running the exec (/usr/local/bin/CubicSDR) resulted in an "app crashed" dialog box popping up, and Terminal reporting seg fault 11. Running with sudo didn't seem to work at all.

This happens with the master branch, and when I replace the six files from the optimizations branch.

Could you post an OS X binary? Thanks!

cjcliffe, 2016-03-17:
@jacobseptember the latest "OSX10.9" build on https://github.com/cjcliffe/CubicSDR/releases/tag/0.1.26-alpha-experiment is based on the #313 branch; give that a go and see if there's any improvement; cheers!

cjcliffe, 2016-03-17:
@jacobseptember hmm, it appears building everything for 10.9 may have caused major slowdown; I'm doing some additional build experiments here to investigate.

cjcliffe, 2016-03-18:
@jacobseptember I've updated 10.9 and 10.0 DMG builds on https://github.com/cjcliffe/CubicSDR/releases/tag/0.1.26-alpha-experiment to fix the performance issue, let me know how it goes when you get the chance.

jacobseptember, 2016-03-18:
All 10.9, 10.10, 10.11 binaries run on my Xeon-based desktop. I only tried 10.10 and 10.11 on my mobile i7-based rMBP, both ran.

Here's what I noticed:

- Gatekeeper and Retina text work great (at least for system-generated text, as opposed to in-window text like the frequency chart and the mode selectors).
- All versions seem to trigger the dGPU on the laptop, taking over from the iGPU. The key in Info.plist is there, and when I added that key to previous versions it stayed on the iGPU, so I assume a newer module causes this. Not a problem per se, just reporting.
- The Soapy dialog box that appears when the program is opened seemed to be slower/buggier on my laptop, at least for the 10.11 build (10.10 was better). I had to close/open that dialog box, and even CubicSDR, a few times before it would recognize my RTL device. Some times it recognized the device but didn't seem to read from it (no waterfall). On my desktop it seems to work perfectly fine.
- On the laptop, it didn't seem to force the CPU to max frequency. On the desktop it _seemed_ to do this at first for all three versions, but after minimizing the window of the 10.11 version the freq. dropped after a while. Surprisingly, un-minimizing the window brought the freq. back up again but NOT to the max, or even close.
- CPU usage, as reported by Activity Monitor, seems to be generally the same as before—low 70% when minimized, 140-160% coming out of minimization. Goes down to to the 70s again after a while though! Very interesting.

In conclusion, it seems that the optimizations generally work as desired. Let me know if there's anything else you want me to look at further. Thanks!

cjcliffe, 2016-03-18:
@jacobseptember thanks for the detailed report; I've been running OSX10.9 build all day at work on an i5 iMac with RTL-SDR and here at home on SDRPlay without a hiccup.  

I also noticed that the 10.11 dialogs and wx in general seems a bit laggy -- I think it might be due to using a 10.9 wxWidgets static build with it.  The 10.9 and 10.10 versions both have wxWidgets builds matching their target but the 10.11 build doesn't include a few of the optimizations and was built with MacPorts libs and 10.9 wxWidgets -- same as what you'd get following the wiki build instructions.

I think the consensus so far is that the 10.9 and 10.10 builds are basically equal in performance so I'm going to target 10.9 as the minimum so that as many users as possible are able to use the release builds.

I'll have some more optimizations for the optimizations_and_fixes branch and a proper release build with all the SDR modules soon -- I'll ping you again when it's available for testing.

Cheers!

cjcliffe, 2016-04-02:
Going to close this one up, check out https://github.com/cjcliffe/CubicSDR/releases/tag/0.1.27-alpha and in addition to the performance improvements you'll find a new menu option called "Reduce CPU Usage" which will clamp the visual rate and keep CubicSDR from chewing up too many cycles.

closed 2016-04-02 by cjcliffe