cjcliffe/CubicSDR #281: No rtl-device showing up in devicelist on newest windows version.

sm0yyq, 2016-02-12:
Hi.

I trie to use the newest windows version on my computer but i cant find my rtl-device in the device list.

It's emty, .. i have installed the nevest zadig drivers as well.

/M

cjcliffe, 2016-02-13:
@sm0yyq thanks for the report -- is this Win32 or Win64 build?  And if you're using Win64 would you be able to check if the Win32 build works for you?

sm0yyq, 2016-02-19:
Hi again.

For my test i used 64 bit variant, but i have also tried the 32 bit variant now.
I cant see any rtl devices now either, but i have got some input's from my soundcard (in the 32 bit version).

/M

cjcliffe, 2016-02-20:
@sm0yyq Are you running Windows 10?  I know you you've installed zadig recently but I just now had what seems like the same problem; I've installed Zadig several times but just now I was unable to see my RTL-SDR device -- downloading latest Zadig and installing again fixed it immediately here.
 
If that doesn't help It's also possible it's packaged incorrectly as I updated several of the Soapy modules and might not realize it here on the build system.  I'll make sure to re-build everything for the next release and get it tested on a system without the build environment present.

sm0yyq, 2016-02-20:
Hi..

Im running Win 7 64bit. A question, do i need to install WinUSB drivers in Zadig to make this work?
I get errors all the time when i trie to install Zadig and WinUSB driver (Timeout errors).
So i use to install LibUSB or what the namn was, ... maybee CubicSDR doesent pic up my dongle if i do soo?

/M

cjcliffe, 2016-02-20:
@sm0yyq I used the default in the Zadig dropdown, I think it's the WinUSB -- also note you need to right-click and run it as Administrator and make sure the 'list all devices' is checked in the menu and RTL device is selected.

sm0yyq, 2016-02-21:
Hi.

I got WinUSB installed at last, ... but no better succes with CubicSDR.. :/

cjcliffe, 2016-03-08:
@sm0yyq not sure what to suggest here; I've tried the latest 0.1.25 build with several RTL dongles here on my windows system without any issues.  Do you have pothosware installed as well by chance?  You might need to add ```-b``` to the command line to force it to use the CubicSDR bundled SoapySDR modules instead of the pothos bundled ones.

sm0yyq, 2016-03-09:
Hi.

Hmm, okey i can test it later.. and no i dont have Pothosware installed (or tested), .. :)

cjcliffe, 2016-03-10:
@sm0yyq you actually might have more luck with the complete Pothos SDR dev environment installer: https://github.com/pothosware/PothosSDR/wiki which now includes CubicSDR anyways :)

Just uninstall any CubicSDR instances first and install the MSVC2015 bundle and it should give you the full set of required libs plus Pothos SDR GUI, GQRX, GNURadio and some other goodies.

sm0yyq, 2016-03-10:
Hi.

I did try install Photos on a fresh Windows 7'en, and i got it to find my dongles after some work.
I also tried after what to install a older version of CubicSDR and it found my dongle.
But i have awfull lag on my system so it doesnt run as smooth it does under Ubuntu.:/

cjcliffe, 2016-03-15:
@sm0yyq I'll close this one up for the moment as next windows GitHub release will be using the latest Pothos dev libs and DLLs anyways -- I'm working on optimizations over at PR #313 and hopefully some of those additions will help improve performance overall soon.

closed 2016-03-15 by cjcliffe