cjcliffe/CubicSDR #282: Add follow option for Rig Control

Toontje, 2016-02-13:
Use case: Two connections to the radio, CubicSDR and HAM Logging program. Both using separate serial ports. When i change the frequency using CubicSDR the HAM Logging software follows the frequency change. When i change the frequency in the logging software CubicSDR doesn't follow.
Following can be implemented by periodically polling the frequency of the radio. This, however, needs to be optional. 
On when i want my panadapter to keep the center frequency the same as the radios VFO.
Off when i want the freedom to scan the bands independent of where i am with my radios VFO. The c key will then center my radio on CubicSDR center frequency.

cjcliffe, 2016-02-13:
@Toontje ok; so I think this covers it,

If I add two toggle options under the rig menu in a sub-menu:

- Frequency Tracking
  - Rig Tracking
  - Rig Control

And regardless of the settings 'C' still sets the rig frequency too?

I think that lets you handle the scenarios above but you can still re-center it at any time.

Toontje, 2016-02-14:
Yes, keep the C alive. It's very useful when quickly changing frequencies.

Toontje, 2016-02-14:
Did you change something in the code or is it just plain luck that CubicSDR is now following the logging program? I pulled down the latest code this morning and don't see anything changed in the menus, but all is working as i like it to work.

cjcliffe, 2016-02-14:
@Toontje not sure; it's supposed to both follow and control by default I assumed you were pitching the use case where they could be toggled off :)

Right now I'm repeatedly polling hamlib for the current frequency; I see there's a call-back option that lets hamlib raise the tracking events directly -- perhaps if I implement that the control will be a bit more consistent.

Analias, 2016-02-15:
I just did a pull from github the changes you implemented for rig control are working really well.  I no longer see CubicSDR aggressively setting the frequency back when I switch bands in WSJT-X.  

The frequency lock with the center frequency is working just as I hoped.  I can create two modems (one offset for PSK, the other for JT65) and both keep their offset from the center frequency.  Switching between bands with WSJT-X allows me now to flip between bands easily and the modems follow nicely.

@Toontje, your original post mentioned using two serial ports. I'd be interested if you could explain further what you are trying to do.  Are you using two radios?  I'm connecting multiple applications to my TS-2000 and having CubicSDR and WSJT-X following and control it.  I'm using rigctrld and pointing both apps at the same network port.  With the changes that Charles made it seems to be working smoothly.  If your building CubicSDR from github, you can pick those changes up now (unless Charles has more changes to make, then I'd wait).

Toontje, 2016-02-16:
So what i am doing is controlling the rig using CubicSDR and using a DX
logging program with CAT control. Since OS X cannot share ports i have to
use two separate ports, one Kenwood USB port for CubicSDR and one Prolific
USB to RS232 for the logging software. Luckily my TS-590 can use the USB
and RS232 port at the same time.

cjcliffe, 2016-02-17:
@Toontje Rig control/follow toggle is now available, also RigThread won't cause as many crashes and if it fails gracefully the "Enable Rig" option will simply un-check itself.

Also I see what @Analias is saying about the rigctld; if you run rigctld with the mode/port/serial options for your rig you can connect as many applications as you'd like via net rigctl -- rigctld maintains a single connection to the rig and exposes it to the network.  Basically I've been doing that here using the 'dummy' rig to connect FLDigi to CubicSDR to test hamlib stuff -- rigctld propagates the dummy rig values to all clients.

Toontje, 2016-02-23:
Can i open this issue with a request to lock the center and LO frequency? Use case: Controlling the rig using CubicSDR and having to press C to continuously link/sync the rig frequency to the center LO frequency.

cjcliffe, 2016-02-23:
@Toontje sure that sounds reasonable, I'll reopen and add that to the list

cjcliffe, 2016-03-01:
@Toontje I've added 'Track Modem' and 'Floating Center' to the Rig menu which provide the rig control case requested plus some combination logic that might be useful.

closed 2016-03-01 by cjcliffe