cjcliffe/CubicSDR #283: SoapySDR remote does not start on Mac OS X

andresv, 2016-02-13:
CubicSDR.0.1.25-alpha.

[image]

Nothing happens when I hit `Start` button. I can see that no connection is made to server.
Server indeed works because it can be opened with `osmocom_fft `:
```
osmocom_fft -a "soapy=0,driver=remote,remote:driver=rtl,remote=10.100.101.2" -s 1024000 -g 20 -v -F
```

cjcliffe, 2016-02-14:
Possible I broke this in a recent update to the Add Remote option; will take another look

cjcliffe, 2016-02-15:
Hmm, appears to be working here on RPi2 but I've already got it showing in the local network section; adding it to remote creates a duplicate that appears to work too -- will try moving it outside my network and see if that changes anything.

andresv, 2016-02-15:
I was using openvpn to connect server and my local machine together, but that should not be the problem and indeed actually CubicSDR is able to connect to server, otherwise there would be no remote devices found.

cjcliffe, 2016-02-15:
ok; I'll try using openvpn in a similar setup here; it's possible that you may need some additional UDP / MTU related openvpn settings.

cjcliffe, 2016-02-17:
@andresv currently running a 250kHz stream from an RTL-SDR on my work computer which I've connected via OpenVPN (TunnelBlick) it's been stable for about 20min so far.

Are there any other conditions I need to reproduce the issue?  I did notice it took about 10-15 seconds to refresh the remote server and see the device properties after clicking the device but after that it seemed to work ok.

andresv, 2016-02-17:
Strange. I can do some more tests on Friday.

andresv, 2016-02-17:
Actually tried just now and indeed connection is only made to server if i hit the `refresh` button, but nothing happens with `start`.

Not sure how to debug it further.

cjcliffe, 2016-02-18:
@andresv it's really strange that the device properties aren't shown when you select the device; that means it was able to see the device but wasn't able to query it.  I'd imagine that's also why it's unable to start.  You should see an additional network connection occur when device selected during querying. 

Make sure SoapyRemote and SoapyRTLSDR / SoapyOsmo are up to date and match the host version just in case something has changed in the protocol or handling.

closed 2016-08-16 by cjcliffe